gazebosim/gz-rendering #589: Adding glutReshapeFunc to examples cause incorrect rendering.

Levi-Armstrong, 2022-03-20:
I noticed that none of the glut examples support resizing. I attempted to updated the mesh viewer example to allow resizing but ended up with incorrect rendering as show below. I assume since none of the examples have a reshape implementation this might be a known issue, but not sure. Am I doing something incorrect?

reshape function
``` c++
void reshapeCB(int w, int h)
{
  imgw = w;
  imgh = h;
  glViewport (0, 0, (GLsizei) w, (GLsizei) h);
}
```

updated display callback
``` c++
void displayCB()
{
#if __APPLE__
  CGLSetCurrentContext(g_context);
#elif _WIN32
#else
  if (g_display)
  {
    glXMakeCurrent(g_display, g_drawable, g_context);
  }
#endif

  // Resize camera 
  auto camera = g_cameras[g_cameraIndex];
  camera->SetImageHeight(imgh);
  camera->SetImageWidth(imgw);
  camera->SetAspectRatio(static_cast<double>(imgw) / static_cast<double>(imgh));
  ir::Image image = camera->CreateImage();
  g_image = std::make_shared<ir::Image>(image);
  camera->Capture(*g_image);

//  g_cameras[g_cameraIndex]->Capture(*g_image);
  handleMouse();

#if __APPLE__
  CGLSetCurrentContext(g_glutContext);
#elif _WIN32
#else
  glXMakeCurrent(g_glutDisplay, g_glutDrawable, g_glutContext);
#endif

  unsigned char *data = g_image->Data<unsigned char>();

  glClearColor(0.5, 0.5, 0.5, 1);
  glClear(GL_COLOR_BUFFER_BIT | GL_DEPTH_BUFFER_BIT);
  glPixelZoom(1, -1);
  glRasterPos2f(-1, 1);
  glDrawPixels(imgw, imgh, GL_RGB, GL_UNSIGNED_BYTE, data);

  glutSwapBuffers();
}
```

[image]

Levi-Armstrong, 2022-03-21:
I thought maybe it was due to using the function `glDrawPixels` which is deprecated and removed from newer version of opengl. This did not fix the issue above but it now segfaults when calling glTexImage2D randomly which makes me think the data extracted from the image return from camera capture is somehow changing.

Original:

``` c++
  glPixelZoom(1, -1);
  glRasterPos2f(-1, 1);
  glDrawPixels(imgw, imgh, GL_RGB, GL_UNSIGNED_BYTE, data);
```

Replaced using Texture and GL_QUAD:

``` c++
  glEnable(GL_TEXTURE_RECTANGLE);
  glViewport (0, 0, (GLsizei) imgw, (GLsizei) imgh);
  glGenTextures (1, &g_texture);
  glBindTexture (GL_TEXTURE_RECTANGLE, g_texture);
  glTexImage2D (GL_TEXTURE_RECTANGLE, 0, GL_RGB, (GLint)imgw, (GLint)imgh, 0, GL_RGB, GL_UNSIGNED_BYTE, data);
  glBegin (GL_QUADS);
  // Bottom left
  glTexCoord2d (imgw, imgh);
  glVertex2f(-1, -1);

  // Top left
  glTexCoord2d (0, imgh);
  glVertex2f (1, -1);

  // Top right
  glTexCoord2d (0, 0);
  glVertex2f (1, 1);

  // Bottom right
  glTexCoord2d (imgw, 0);
  glVertex2f (-1, 1);
  glEnd();
  glDisable(GL_TEXTURE_RECTANGLE);
```

Levi-Armstrong, 2022-03-21:
When it segfaults it outputs the following:
``` bash
RTTI symbol not found for class 'ignition::rendering::v6::OgreCamera'
RTTI symbol not found for class 'ignition::rendering::v6::OgreCamera'
RTTI symbol not found for class 'ignition::rendering::v6::OgreCamera'
```

Levi-Armstrong, 2022-03-21:
It turns out it is an alignment issues. If you add the following lines before loading the texture everything looks good and no segfault.

``` c++
  glPixelStorei(GL_UNPACK_ALIGNMENT, 1);
  glPixelStorei(GL_UNPACK_ROW_LENGTH, 0);
  glPixelStorei(GL_UNPACK_SKIP_PIXELS, 0);
  glPixelStorei(GL_UNPACK_SKIP_ROWS, 0);
```

``` c++
  glClearColor(0.5, 0.5, 0.5, 1);
  glClear(GL_COLOR_BUFFER_BIT | GL_DEPTH_BUFFER_BIT);

  glEnable(GL_TEXTURE_RECTANGLE);
  glViewport (0, 0, (GLsizei) imgw, (GLsizei) imgh);
  glDeleteTextures(1, &g_texture);
  glGenTextures (1, &g_texture);
  glBindTexture (GL_TEXTURE_RECTANGLE, g_texture);
  glPixelStorei(GL_UNPACK_ALIGNMENT, 1);
  glPixelStorei(GL_UNPACK_ROW_LENGTH, 0);
  glPixelStorei(GL_UNPACK_SKIP_PIXELS, 0);
  glPixelStorei(GL_UNPACK_SKIP_ROWS, 0);
  glTexImage2D (GL_TEXTURE_RECTANGLE, 0, GL_RGB, (GLint)imgw, (GLint)imgh, 0, GL_RGB, GL_UNSIGNED_BYTE, data);
  glBegin (GL_QUADS);
  // Bottom left
  glTexCoord2d (imgw, imgh);
  glVertex2f(-1, -1);

  // Top left
  glTexCoord2d (0, imgh);
  glVertex2f (1, -1);

  // Top right
  glTexCoord2d (0, 0);
  glVertex2f (1, 1);

  // Bottom right
  glTexCoord2d (imgw, 0);
  glVertex2f (-1, 1);
  glEnd();
  glDisable(GL_TEXTURE_RECTANGLE);
```

darksylinc, 2022-04-30:
You're not calling `Camera::SetImageWidth` & `SetImageHeight`.

Though I'm not sure that's enough. The problem with the GLUT samples is that they're mixing OgreNext's Core Profile (good) with Compatibility Profile (bad).

For example `glDrawPixels` is deprecated and can cause a lot of problems. It's a miracle it works actually.

After adding the necessary changes to Ignition-Rendering I've created [a sample based on SDL2](https://github.com/darksylinc/ign-rendering/blob/matias-2.3-global-illumination/examples/global_illumination/GlutWindow.cc) instead which supports resizing.

It hasn't yet been merged with main; and the goal is to slowly move these samples to SDL2.

Levi-Armstrong, 2022-05-02:
> You're not calling Camera::SetImageWidth & SetImageHeight.
@darksylinc In my original post I am calling camera.SetImageWidth and camera.SetImageHeight or are you referring to something else?

Thanks for the information and I will take a look at the SDL2 examples.