gazebosim/gz-rendering #597: Fix PlanarReflection support in buildsystem (used by Heightmap)

j-rivero, 2022-03-30:
## Environment
* OS Version: 20.04 / 22.04
* Source or binary build? Source builds when creating packages in Ubuntu Focal and Jammy

## Description
When creating the Ubuntu Jammy packages using ogre-next from Ubuntu repositories, the dependencies of the package brought into the system the `libogrenextplanarreflections2.2.5` package. While Ignition does not use the PlanarReflection component directly, the heightmap feature uses it (mainly code in directory src/terrain/Terra/src/Hlms). The use of PlanarReflection is [conditionally controlled in the source code](https://github.com/ignitionrobotics/ign-rendering/blob/ign-rendering6/ogre2/src/terrain/Terra/src/Hlms/OgreHlmsTerra.cpp#L57-L59) but the buildsystem seems not to have any reference to PlanarReflection.

The Jammy package builds corresponding to version 6.2.1 (and probably lower) displays a problem with missing symbols found:
```
dpkg-shlibdeps: warning: symbol _ZNK4Ogre17PlanarReflections26_isUpdatingRenderablesHlmsEv used by debian/libignition-rendering6-ogre2/usr/lib/x86_64-linux-gnu/libignition-rendering6-ogre2.so.6.2.2~pre2 found in none of the libraries
dpkg-shlibdeps: warning: symbol _ZNK4Ogre17PlanarReflections20hasPlanarReflectionsEPKNS_10RenderableE used by debian/libignition-rendering6-ogre2/usr/lib/x86_64-linux-gnu/libignition-rendering6-ogre2.so.6.2.2~pre2 found in none of the libraries
dpkg-shlibdeps: warning: symbol _ZN4Ogre17PlanarReflections37_notifyRenderableFlushedHlmsDatablockEPNS_10RenderableE used by debian/libignition-rendering6-ogre2/usr/lib/x86_64-linux-gnu/libignition-rendering6-ogre2.so.6.2.2~pre2 found in none of the libraries
dpkg-shlibdeps: warning: symbol XDestroyWindow used by debian/libignition-rendering6-ogre2/usr/lib/x86_64-linux-gnu/libignition-rendering6-ogre2.so.6.2.2~pre2 found in none of the libraries
dpkg-shlibdeps: warning: symbol _ZNK4Ogre17PlanarReflections10getTextureEh used by debian/libignition-rendering6-ogre2/usr/lib/x86_64-linux-gnu/libignition-rendering6-ogre2.so.6.2.2~pre2 found in none of the libraries
dpkg-shlibdeps: warning: debian/libignition-rendering6-ogre2/usr/lib/x86_64-linux-gnu/ign-rendering-6/engine-plugins/libignition-rendering6-ogre2.so.6.2.2~pre2 contains an unresolvable reference to symbol _ZN4Ogre17PlanarReflections37_notifyRenderableFlushedHlmsDatablockEPNS_10RenderableE: it's probably a plugin
```
I was not able to make the simulator to crash by running on Jammy the example of `heightmap.sdf`, not sure if PlanarReflections is being loaded as a plugin or I did not run the right code to make it crash.

The current [buildsystem does not include PlanarReflections](https://github.com/ignitionrobotics/ign-rendering/blob/ign-rendering6/CMakeLists.txt#L106-L109) when looking for Ogre2 components.

j-rivero, 2022-04-05:
Support in ign-cmake to handle the PlanarReflections component https://github.com/ignitionrobotics/ign-cmake/pull/224

darksylinc, 2022-04-30:
One possible solution is what EmptyProject uses:

```cmake
file( READ "${OGRE_BINARIES}/include/OgreBuildSettings.h" OGRE_BUILD_SETTINGS_STR )
string( FIND "${OGRE_BUILD_SETTINGS_STR}" "#define OGRE_BUILD_COMPONENT_PLANAR_REFLECTIONS" IS_PLANAR_REFL )
if( NOT IS_PLANAR_REFL EQUAL -1 )
    # Planar Reflections are being used
endif()
```

It basically looks for OgreBuildSettings.h, parses it and checks for the presence of `#define OGRE_BUILD_COMPONENT_PLANAR_REFLECTIONS`