gazebosim/gz-rendering #598: Difficult to orbit around heightmaps

chapulina, 2022-03-30:
## Environment
* OS Version:  Ubuntu Focal
* Source or binary build? Source, `main` or `ign-rendering6`
  
  

## Description
* Expected behavior:  Using the mouse to orbit around a heightmap should be as easy as orbiting around other meshes.
* Actual behavior:  It looks like the ray doesn't intersect the heightmap and I'm orbiting around an arbitrary point in space. 

See how it moves much more when orbiting the sphere than on the heightmap:

[image]

This becomes more problematic the further you are. Here's an example with a large heightmap. Orbiting the grid is much easier:

[image]

## Steps to reproduce

1. `ign gazebo heightmap.sdf`
2. Orbit around the heightmap and the other models

## Additional context

I remember we had the same issue on Gazebo classic a while back:

* Issue: https://github.com/osrf/gazebo/issues/2049
* PR diff: https://github.com/osrf/gazebo/compare/482a93b0877be937651ff85413ece0ca7718b23a..9a92f437ef54261bbdbd09243a8a7b7fdb0fa95b

Kakcalu13, 2022-04-19:
is this still issue? Seems OK for me now unless I mistook the bug? I am on Fortress 6.9.0 version (ignore the render issue)

[image]

chapulina, 2022-04-22:
I think it works reasonably when you're close to the heightmap, but the further away, the worse it gets. My 2nd heightmap spans multiple kilometers, and if I try to orbit it from far away, it's very difficult.