gazebosim/gz-rendering #600: ignition cmake  /usr/bin/ld: cannot find -lpthreads

aradhyamathur, 2022-04-01:
## Environment
* OS Version: Ubuntu 20.04 
* Source 
 commit f043b4fc8aa234d95b570b6ad61db0c8592be007 (HEAD -> ign-rendering6, origin/ign-rendering6, origin/HEAD)

## Steps to reproduce
1. Create build directory `mkdir build`
2. `cd build`
3. `mkdir install`
4. `cmake .. -DCMAKE_INSTALL_PREFIX=./install`

## Output
CMake Error Log
```
Run Build Command(s):/usr/bin/make cmTC_4322d/fast && /usr/bin/make -f CMakeFiles/cmTC_4322d.dir/build.make CMakeFiles/cmTC_4322d.dir/build
make[1]: Entering directory '/home/thor/gsoc/ign-rendering/build/CMakeFiles/CMakeTmp'
Building C object CMakeFiles/cmTC_4322d.dir/src.c.o
/usr/bin/cc   -fPIC -DCMAKE_HAVE_LIBC_PTHREAD   -o CMakeFiles/cmTC_4322d.dir/src.c.o   -c /home/thor/gsoc/ign-rendering/build/CMakeFiles/CMakeTmp/src.c
Linking C executable cmTC_4322d
/usr/bin/cmake -E cmake_link_script CMakeFiles/cmTC_4322d.dir/link.txt --verbose=1
/usr/bin/cc -fPIC -DCMAKE_HAVE_LIBC_PTHREAD    CMakeFiles/cmTC_4322d.dir/src.c.o  -o cmTC_4322d 
/usr/bin/ld: CMakeFiles/cmTC_4322d.dir/src.c.o: in function `main':
src.c:(.text+0x46): undefined reference to `pthread_create'
/usr/bin/ld: src.c:(.text+0x52): undefined reference to `pthread_detach'
/usr/bin/ld: src.c:(.text+0x63): undefined reference to `pthread_join'
collect2: error: ld returned 1 exit status
make[1]: *** [CMakeFiles/cmTC_4322d.dir/build.make:87: cmTC_4322d] Error 1
make[1]: Leaving directory '/home/thor/gsoc/ign-rendering/build/CMakeFiles/CMakeTmp'
make: *** [Makefile:121: cmTC_4322d/fast] Error 2

Source file was:
#include <pthread.h>

void* test_func(void* data)
{
  return data;
}

int main(void)
{
  pthread_t thread;
  pthread_create(&thread, NULL, test_func, NULL);
  pthread_detach(thread);
  pthread_join(thread, NULL);
  pthread_atfork(NULL, NULL, NULL);
  pthread_exit(NULL);

  return 0;
}
```

```
Run Build Command(s):/usr/bin/make cmTC_dbb13/fast && /usr/bin/make -f CMakeFiles/cmTC_dbb13.dir/build.make CMakeFiles/cmTC_dbb13.dir/build
make[1]: Entering directory '/home/thor/gsoc/ign-rendering/build/CMakeFiles/CMakeTmp'
Building C object CMakeFiles/cmTC_dbb13.dir/CheckFunctionExists.c.o
/usr/bin/cc   -fPIC -DCHECK_FUNCTION_EXISTS=pthread_create   -o CMakeFiles/cmTC_dbb13.dir/CheckFunctionExists.c.o   -c /usr/share/cmake-3.16/Modules/CheckFunctionExists.c
Linking C executable cmTC_dbb13
/usr/bin/cmake -E cmake_link_script CMakeFiles/cmTC_dbb13.dir/link.txt --verbose=1
/usr/bin/cc -fPIC -DCHECK_FUNCTION_EXISTS=pthread_create    CMakeFiles/cmTC_dbb13.dir/CheckFunctionExists.c.o  -o cmTC_dbb13  -lpthreads 
/usr/bin/ld: cannot find -lpthreads
collect2: error: ld returned 1 exit status
make[1]: *** [CMakeFiles/cmTC_dbb13.dir/build.make:87: cmTC_dbb13] Error 1
make[1]: Leaving directory '/home/thor/gsoc/ign-rendering/build/CMakeFiles/CMakeTmp'
make: *** [Makefile:121: cmTC_dbb13/fast] Error 2
```

Complete CMake output 
```
cmake .. -DCMAKE_INSTALL_PREFIX=./install
-- The C compiler identification is GNU 8.4.0
[... 132 lines cut ...]

CMake Error at /usr/share/cmake/ignition-cmake2/cmake2/IgnConfigureBuild.cmake:59 (message):
  -- BUILD ERRORS: These must be resolved before compiling.
Call Stack (most recent call first):
  CMakeLists.txt:167 (ign_configure_build)

CMake Error at /usr/share/cmake/ignition-cmake2/cmake2/IgnConfigureBuild.cmake:61 (message):
  -- 	Missing dependency [ignition-math6] (Components: eigen3)
Call Stack (most recent call first):
  CMakeLists.txt:167 (ign_configure_build)

CMake Error at /usr/share/cmake/ignition-cmake2/cmake2/IgnConfigureBuild.cmake:63 (message):
  -- END BUILD ERRORS

Call Stack (most recent call first):
  CMakeLists.txt:167 (ign_configure_build)

CMake Error at /usr/share/cmake/ignition-cmake2/cmake2/IgnConfigureBuild.cmake:68 (message):
  Errors encountered in build.  Please see BUILD ERRORS above.
Call Stack (most recent call first):
  CMakeLists.txt:167 (ign_configure_build)

-- Configuring incomplete, errors occurred!
See also "/home/thor/gsoc/ign-rendering/build/CMakeFiles/CMakeOutput.log".
See also "/home/thor/gsoc/ign-rendering/build/CMakeFiles/CMakeError.log".```

aradhyamathur, 2022-04-01:
Upon further investigation it seems like the problem arises due to cmake using -lpthread, if I compile the code mentioned above in source file in cmake error logs and  compile it with -pthread , it compiles fine. However, I am unable to find the option to make cmake use -pthread instead of -lpthread.

aradhyamathur, 2022-04-01:
set(THREADS_PREFER_PTHREAD_FLAG ON) adding this to the CMakeLists.txt does reduce the number of errors produced from lpthreads however still the sample file leads to the same error.

iche033, 2022-04-02:
the main build error seems to be:

```
  Could not find a package configuration file provided by
  "ignition-math6-eigen3" (requested version 6.10.0) with any of the
  following names:
```

you'll need the eigen3 component from ign-math. You can take a look here for installation instructions:

https://ignitionrobotics.org/api/math/6.8/install.html

aradhyamathur, 2022-04-02:
yes indeed it got resolved with this. Thanks.

closed 2022-04-02 by aradhyamathur

osrf-triage, 2022-04-04:
This issue has been mentioned on **Gazebo Community**. There might be relevant details there:

https://community.gazebosim.org/t/gsoc-2022-texture-domain-randomization/1344/6