gazebosim/gz-rendering #615: Leaks from Shaders on shutdown

darksylinc, 2022-05-01:
## Environment
* OS Version: Ubuntu 20.04
* Source or binary build?
  main

## Steps to reproduce

Running `INTEGRATION_camera` in the Vulkan branch reveals the following leaks:

```
ERROR: [Validation] Code 0 : Validation Error: [ VUID-vkDestroyDevice-device-00378 ] Object 0: handle = 0x55f71e464fa0, type = VK_OBJECT_TYPE_DEVICE; Object 1: handle = 0x74d977000000004a, name = HeatSignatureVS_VK, type = VK_OBJECT_TYPE_SHADER_MODULE; | MessageID = 0x71500fba | OBJ ERROR : For VkDevice 0x55f71e464fa0[], VkShaderModule 0x74d977000000004a[HeatSignatureVS_VK] has not been destroyed. The Vulkan spec states: All child objects created on device must have been destroyed prior to destroying device (https://vulkan.lunarg.com/doc/view/1.2.198.1/linux/1.2-extensions/vkspec.html#VUID-vkDestroyDevice-device-00378)
16:27:55: ERROR: [Validation] Code 0 : Validation Error: [ VUID-vkDestroyDevice-device-00378 ] Object 0: handle = 0x55f71e464fa0, type = VK_OBJECT_TYPE_DEVICE; Object 1: handle = 0x67022e000000004b, name = HeatSignatureFS_VK, type = VK_OBJECT_TYPE_SHADER_MODULE; | MessageID = 0x71500fba | OBJ ERROR : For VkDevice 0x55f71e464fa0[], VkShaderModule 0x67022e000000004b[HeatSignatureFS_VK] has not been destroyed. The Vulkan spec states: All child objects created on device must have been destroyed prior to destroying device (https://vulkan.lunarg.com/doc/view/1.2.198.1/linux/1.2-extensions/vkspec.html#VUID-vkDestroyDevice-device-00378)
16:27:55: ERROR: [Validation] Code 0 : Validation Error: [ VUID-vkDestroyDevice-device-00378 ] Object 0: handle = 0x55f71e464fa0, type = VK_OBJECT_TYPE_DEVICE; Object 1: handle = 0x74ea280000000054, name = selection_buffer_vs_VK, type = VK_OBJECT_TYPE_SHADER_MODULE; | MessageID = 0x71500fba | OBJ ERROR : For VkDevice 0x55f71e464fa0[], VkShaderModule 0x74ea280000000054[selection_buffer_vs_VK] has not been destroyed. The Vulkan spec states: All child objects created on device must have been destroyed prior to destroying device (https://vulkan.lunarg.com/doc/view/1.2.198.1/linux/1.2-extensions/vkspec.html#VUID-vkDestroyDevice-device-00378)
16:27:55: ERROR: [Validation] Code 0 : Validation Error: [ VUID-vkDestroyDevice-device-00378 ] Object 0: handle = 0x55f71e464fa0, type = VK_OBJECT_TYPE_DEVICE; Object 1: handle = 0x6871490000000055, name = selection_buffer_fs_VK, type = VK_OBJECT_TYPE_SHADER_MODULE; | MessageID = 0x71500fba | OBJ ERROR : For VkDevice 0x55f71e464fa0[], VkShaderModule 0x6871490000000055[selection_buffer_fs_VK] has not been destroyed. The Vulkan spec states: All child objects created on device must have been destroyed prior to destroying device (https://vulkan.lunarg.com/doc/view/1.2.198.1/linux/1.2-extensions/vkspec.html#VUID-vkDestroyDevice-device-00378)
16:27:55: ERROR: [Validation] Code 0 : Validation Error: [ VUID-vkDestroyDevice-device-00378 ] Object 0: handle = 0x55f71e464fa0, type = VK_OBJECT_TYPE_DEVICE; Object 1: handle = 0x603cd90000000044, name = ThermalCameraVS_VK, type = VK_OBJECT_TYPE_SHADER_MODULE; | MessageID = 0x71500fba | OBJ ERROR : For VkDevice 0x55f71e464fa0[], VkShaderModule 0x603cd90000000044[ThermalCameraVS_VK] has not been destroyed. The Vulkan spec states: All child objects created on device must have been destroyed prior to destroying device (https://vulkan.lunarg.com/doc/view/1.2.198.1/linux/1.2-extensions/vkspec.html#VUID-vkDestroyDevice-device-00378)
16:27:55: ERROR: [Validation] Code 0 : Validation Error: [ VUID-vkDestroyDevice-device-00378 ] Object 0: handle = 0x55f71e464fa0, type = VK_OBJECT_TYPE_DEVICE; Object 1: handle = 0x8cc1d80000000045, name = ThermalCameraFS_VK, type = VK_OBJECT_TYPE_SHADER_MODULE; | MessageID = 0x71500fba | OBJ ERROR : For VkDevice 0x55f71e464fa0[], VkShaderModule 0x8cc1d80000000045[ThermalCameraFS_VK] has not been destroyed. The Vulkan spec states: All child objects created on device must have been destroyed prior to destroying device (https://vulkan.lunarg.com/doc/view/1.2.198.1/linux/1.2-extensions/vkspec.html#VUID-vkDestroyDevice-device-00378)
```

Although it would look like Vulkan errors; these errors are very likely happening because Ignition is holding to a SharedPtr after OgreNext has been shutdown; and it would affects all APIs (it's just happens that Vulkan layers caught it).

These modules all come from low level shaders, so it's most likely coming from either `Ogre2Material` or the Cameras (e.g. ThermalCamera, etc)

I will track this bug myself. I'm opening the ticket so I don't forget.

darksylinc, 2022-05-01:
@iche033 I tracked down the problem to the following in camera.cc:

## Old code:

```cpp
void CameraTest::ShaderSelection(const std::string &_renderEngine)
{
  auto thermalCamera = scene->CreateThermalCamera("ThermalCamera");
  //...
  engine->DestroyScene(scene);
  rendering::unloadEngine(engine->Name());
}
```

## Fixed code:

```cpp
void CameraTest::ShaderSelection(const std::string &_renderEngine)
{
  {
    auto thermalCamera = scene->CreateThermalCamera("ThermalCamera");
    //...
    root->RemoveChild(thermalCamera);
    engine->DestroyScene(scene);
  }
  rendering::unloadEngine(engine->Name());
}
```

The extra scope is something I was expecting (i.e. smart_ptr should be reset before `rendering::unloadEngine`). However I noticed the call `root->RemoveChild(thermalCamera);` is required.

This looks like a bug; as I think this is something that should be happening as part of `DestroyScene()` call (because the Root Visual afaik belongs to the Scene)

@iche033 : Is this behaviour a bug or intended?

darksylinc, 2022-05-01:
Btw this can be easily debugged with:

`fprintf( stderr, "%i\n", thermalCamera.use_count() );`

After `engine->DestroyScene(scene);` the value of `thermalCamera.use_count()` should be 1.
But when `root->RemoveChild(thermalCamera);` isn't called, the count is 2 instead.

iche033, 2022-05-03:
Nope that's not intended. Definitely an issue, which should hopefully be fixed by https://github.com/ignitionrobotics/ign-rendering/pull/617