gazebosim/gz-rendering #621: New C4275 warnings on Windows from OgreUTFString

chapulina, 2022-05-06:
## Environment
* OS Version:  Windows
* Source or binary build? Source, `main` branch
  
  

## Description
* Expected behavior:  No warnings on CI
* Actual behavior:  100 warnings on CI

## Steps to reproduce

Run Windows CI.

## Output

See for example:

https://build.osrfoundation.org/job/ign_rendering-pr-win/2552/msbuild/

```
non dll-interface class 'std::runtime_error' used as base for dll-interface class 'Ogre::UTFString::invalid_data' (compiling source file C:\Jenkins\workspace\ign_rendering-pr-win\ws\ign-rendering\ogre\src\OgreConversions.cc)
```

chapulina, 2022-07-29:
* #622 

 via

* #617 

fixed the warnings for Fortress, but not for Citadel - that's been yellow since May:

https://build.osrfoundation.org/job/ign_rendering-ign-3-win/

I've removed the Windows check requirement for `ign-rendering3` and updated 

* https://github.com/gazebo-tooling/release-tools/issues/398

@Blast545 / @Crola1702 - this would be a good one to tackle soon

Crola1702, 2022-08-10:
@mjcarroll I'm looking into this problem. I was checking this https://github.com/gazebosim/gz-rendering/pull/617 PR and thinking about backporting just macros in the files. 

Are you looking into this? Or, can I self assign this for the moment?

mjcarroll, 2022-08-10:
I'm not currently looking at it, but it's in my queue.   If you want to grab it, I would appreciate.

I think that the macros are exactly what is needed here.

Crola1702, 2022-08-17:
This was fixed with: https://github.com/gazebosim/gz-rendering/pull/707
It's working here: https://build.osrfoundation.org/job/ign_rendering-ign-3-win/80/msbuild/folder.702720613/#categoryContent
We can close this now.

closed 2022-08-17 by Blast545