gazebosim/gz-rendering #628: failure to build with optix

xlla, 2022-05-17:
## Environment
* OS Version: macOS 10.13.6
* Source or binary build?
  source, main, ff16689ff6c7c96a7fb58b57e3fa9f11e844c283

<details>
  
```
In file included from /Users/xlla/git/gz-rendering/optix/src/OptixSphere.cc:17:
In file included from /Users/xlla/git/gz-rendering/optix/include/ignition/rendering/optix/OptixSphere.hh:18:
In file included from /Users/xlla/git/gz-rendering/optix/include/gz/rendering/optix/OptixSphere.hh:20:
In file included from /Users/xlla/git/gz-rendering/optix/include/gz/rendering/optix/OptixPrimitive.hh:21:
In file included from /Users/xlla/git/gz-rendering/optix/include/gz/rendering/optix/OptixGeometry.hh:21:
In file included from /Users/xlla/git/gz-rendering/optix/include/gz/rendering/optix/OptixIncludes.hh:21:
In file included from /usr/local/optix/include/optixu/optixpp.h:35:
In file included from /usr/local/optix/include/optixu/optixu_math.h:52:
In file included from /usr/local/optix/include/optixu/optixu_math_namespace.h:41:
In file included from /usr/local/optix/include/optixu/../internal/optix_datatypes.h:33:
/usr/local/cuda/include/host_defines.h:54:2: warning: "host_defines.h is an internal header file and must not be used directly. This file will be removed in a future CUDA release. Please use cuda_runtime_api.h
      or cuda_runtime.h instead." [-W#warnings]
#warning "host_defines.h is an internal header file and must not be used directly.  This file will be removed in a future CUDA release.  Please use cuda_runtime_api.h or cuda_runtime.h instead."
 ^
/Users/xlla/git/gz-rendering/optix/src/OptixScene.cc:206:38: error: allocating an object of abstract class type 'ignition::rendering::v7::OptixDirectionalLight'
  OptixDirectionalLightPtr light(new OptixDirectionalLight);
                                     ^
/Users/xlla/git/gz-rendering/include/gz/rendering/Light.hh:112:30: note: unimplemented pure virtual method 'Intensity' in 'OptixDirectionalLight'
      public: virtual double Intensity() const = 0;
                             ^
/Users/xlla/git/gz-rendering/include/gz/rendering/Light.hh:116:28: note: unimplemented pure virtual method 'SetIntensity' in 'OptixDirectionalLight'
      public: virtual void SetIntensity(double _intensity) = 0;
                           ^
/Users/xlla/git/gz-rendering/optix/src/OptixScene.cc:215:32: error: allocating an object of abstract class type 'ignition::rendering::v7::OptixPointLight'
  OptixPointLightPtr light(new OptixPointLight);
                               ^
/Users/xlla/git/gz-rendering/include/gz/rendering/Light.hh:112:30: note: unimplemented pure virtual method 'Intensity' in 'OptixPointLight'
      public: virtual double Intensity() const = 0;
                             ^
/Users/xlla/git/gz-rendering/include/gz/rendering/Light.hh:116:28: note: unimplemented pure virtual method 'SetIntensity' in 'OptixPointLight'
      public: virtual void SetIntensity(double _intensity) = 0;
                           ^
/Users/xlla/git/gz-rendering/optix/src/OptixScene.cc:224:31: error: allocating an object of abstract class type 'ignition::rendering::v7::OptixSpotLight'
  OptixSpotLightPtr light(new OptixSpotLight);
                              ^
/Users/xlla/git/gz-rendering/include/gz/rendering/Light.hh:112:30: note: unimplemented pure virtual method 'Intensity' in 'OptixSpotLight'
      public: virtual double Intensity() const = 0;
                             ^
/Users/xlla/git/gz-rendering/include/gz/rendering/Light.hh:116:28: note: unimplemented pure virtual method 'SetIntensity' in 'OptixSpotLight'
      public: virtual void SetIntensity(double _intensity) = 0;
                           ^
In file included from /Users/xlla/git/gz-rendering/optix/src/OptixRenderEngine.cc:26:

```

</details>

## Description
* Expected behavior: build libs
* Actual behavior:  failure

<pre>
make[2]: Leaving directory '/Users/xlla/git/gz-rendering/build'
CMakeFiles/Makefile2:663: recipe for target 'optix/src/CMakeFiles/ignition-rendering7-optix.dir/all' failed
make[1]: *** [optix/src/CMakeFiles/ignition-rendering7-optix.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
</pre>

## Steps to reproduce
1. clone the source
2. checkout origin/main
3. mkdir build && cd build
4. cmake 

<pre>
cmake \
.. \
-DCMAKE_INSTALL_PREFIX=/usr/local/Cellar/ignition-rendering7/6.999.999~0~20220414 \
-DCMAKE_INSTALL_LIBDIR=lib \
-DCMAKE_BUILD_TYPE=Release \
-DCMAKE_FIND_FRAMEWORK=LAST \
-DCMAKE_VERBOSE_MAKEFILE=ON \
-Wno-dev \
-DBUILD_TESTING=OFF \
-DBUILD_TESTING=Off \
-DCMAKE_INSTALL_RPATH=@loader_path/../lib \
-DCMAKE_OSX_DEPLOYMENT_TARGET=10.14
</pre>

## Output

xlla, 2022-05-17:
after add those two blank implement functions, more errors occur:

<pre>
/Users/xlla/git/gz-rendering/optix/include/gz/rendering/optix/OptixLightManager.hh:42:61: note: passing argument to parameter '_light' here
      public: virtual void AddPointLight(OptixPointLightPtr _light);
                                                            ^
/Users/xlla/git/gz-rendering/optix/src/OptixLight.cc:243:16: error: no member named 'data' in 'ignition::rendering::v7::OptixPointLight'
  return this->data.common;
         ~~~~  ^
/Users/xlla/git/gz-rendering/optix/src/OptixLight.cc:249:16: error: no member named 'data' in 'ignition::rendering::v7::OptixPointLight'
  return this->data.common;
         ~~~~  ^
/Users/xlla/git/gz-rendering/optix/src/OptixLight.cc:253:37: error: out-of-line definition of 'SharedThis' does not match any declaration in 'ignition::rendering::v7::OptixPointLight'
OptixPointLightPtr OptixPointLight::SharedThis()
                                    ^~~~~~~~~~
</pre>

<pre>
In file included from /Users/xlla/git/gz-rendering/optix/src/OptixRenderEngine.cc:26:
In file included from /Users/xlla/git/gz-rendering/optix/include/ignition/rendering/optix/OptixRenderEngine.hh:18:
/Users/xlla/git/gz-rendering/optix/include/gz/rendering/optix/OptixRenderEngine.hh:77:31: warning: 'InitImpl' overrides a member function but is not marked 'override' [-Winconsistent-missing-override]
      protected: virtual bool InitImpl();
                              ^
/Users/xlla/git/gz-rendering/include/gz/rendering/base/BaseRenderEngine.hh:115:31: note: overridden virtual function is here
      protected: virtual bool InitImpl() = 0;
                              ^
/Users/xlla/git/gz-rendering/optix/src/OptixRenderEngine.cc:111:34: error: allocating an object of abstract class type 'ignition::rendering::v7::OptixScene'
  auto scene = OptixScenePtr(new OptixScene(_id, _name));
                                 ^
/Users/xlla/git/gz-rendering/include/gz/rendering/base/BaseScene.hh:267:39: note: unimplemented pure virtual method 'CreateCOMVisualImpl' in 'OptixScene'
      protected: virtual COMVisualPtr CreateCOMVisualImpl(unsigned int _id,
                                      ^
/Users/xlla/git/gz-rendering/include/gz/rendering/base/BaseScene.hh:274:43: note: unimplemented pure virtual method 'CreateInertiaVisualImpl' in 'OptixScene'
      protected: virtual InertiaVisualPtr CreateInertiaVisualImpl(
                                          ^
/Users/xlla/git/gz-rendering/include/gz/rendering/base/BaseScene.hh:281:41: note: unimplemented pure virtual method 'CreateJointVisualImpl' in 'OptixScene'
      protected: virtual JointVisualPtr CreateJointVisualImpl(unsigned int _id,
                                        ^
/Users/xlla/git/gz-rendering/include/gz/rendering/base/BaseScene.hh:288:41: note: unimplemented pure virtual method 'CreateLightVisualImpl' in 'OptixScene'
      protected: virtual LightVisualPtr CreateLightVisualImpl(unsigned int _id,
                                        ^
/Users/xlla/git/gz-rendering/include/gz/rendering/base/BaseScene.hh:748:37: note: unimplemented pure virtual method 'CreateCapsuleImpl' in 'OptixScene'
      protected: virtual CapsulePtr CreateCapsuleImpl(unsigned int _id,
                                    ^
/Users/xlla/git/gz-rendering/include/gz/rendering/base/BaseScene.hh:762:36: note: unimplemented pure virtual method 'CreateMarkerImpl' in 'OptixScene'
      protected: virtual MarkerPtr CreateMarkerImpl(unsigned int _id,
                                   ^
/Users/xlla/git/gz-rendering/include/gz/rendering/base/BaseScene.hh:769:41: note: unimplemented pure virtual method 'CreateLidarVisualImpl' in 'OptixScene'
      protected: virtual LidarVisualPtr CreateLidarVisualImpl(unsigned int _id,
                                        ^
/Users/xlla/git/gz-rendering/include/gz/rendering/base/BaseScene.hh:777:39: note: unimplemented pure virtual method 'CreateHeightmapImpl' in 'OptixScene'
      protected: virtual HeightmapPtr CreateHeightmapImpl(unsigned int _id,
                                      ^
/Users/xlla/git/gz-rendering/include/gz/rendering/base/BaseScene.hh:785:37: note: unimplemented pure virtual method 'CreateWireBoxImpl' in 'OptixScene'
      protected: virtual WireBoxPtr CreateWireBoxImpl(unsigned int _id,
                                    ^
52 warnings generated.
</pre>

mjcarroll, 2022-05-27:
The optix implementation hasn't been kept in sync with the `ogre` and `ogre2` implementations.  It will likely need some work to bring up-to-date.  Any pull requests in this direction would be greatly appreciated.