gazebosim/gz-rendering #641: Enumerating all Visuals/Sensors incurs O(N²) behavior

darksylinc, 2022-06-05:
## Environment
* OS Version: Ubuntu 20.04
* Source or binary build?
  main branch, but probably all are affected

## Description

So I needed to run through all the sensors (note: this applies to Visuals as well). And I noticed the only way I could do that was by doing:

```cpp
const unsigned int sensorCount = scene->SensorCount();
for (int i = 0; i < sensorCount; ++i)
{
  auto sensor = scene->SensorByIndex(i);
}
```

Being curious that I am, I went what happens inside `SensorByIndex`.

This lead me to:

```cpp
template <class T, class U>
typename BaseStore<T, U>::TPtr
BaseStore<T, U>::GetByIndex(unsigned int _index) const
{
  return this->DerivedByIndex(_index);
}
```

Which landed me here:

```cpp
template <class T, class U>
typename BaseStore<T, U>::ConstUIter
BaseStore<T, U>::ConstIterByIndex(unsigned int _index) const
{
  if (_index >= this->Size())
  {
    ignerr << "Invalid index: " << _index << std::endl;
    return this->store.end();
  }

  auto iter = this->store.begin();
  std::advance(iter, _index);
  return iter;
}
```

The problem is that `this->store` is actually a `std::map<std::string, UPtr>`

This results in two issues:

1. There is no guarantee this->Size() is O(1) (although most stdlib implementations usually make it so by caching the size). Thus every call could potentially end up being O(N). If it gets called N times, then it's O(N²).
2. `std::advance` takes O(N). So the first time call will be O(1), second call will be O(2), and so on. This results in 1 + 2 + ... + N which is `N * (N-1) / 2` this is effectively O(N²)

The container where one could access the iterators is protected (i.e. `protected: virtual SensorStorePtr Sensors() const = 0;`) thus it appears to be impossible for Ignition users to enumerate all sensors/visuals without incurring O(N²)

* Expected behavior: Iterating through all objects should be O(N)
* Actual behavior: Iterating through all objects ends up being O(N²)

## Steps to reproduce

Just run:

```cpp
const unsigned int sensorCount = scene->SensorCount();
for (int i = 0; i < sensorCount; ++i)
{
  auto sensor = scene->SensorByIndex(i);
}
```

Possible solutions are to expose Sensors()/Visuals() or to use a shadow std::vector (which may have other trade-offs)

## Output

There's no output. This is in algorithmic complexity discussion.

iche033, 2022-06-06:
We could expose the `Sensors()` API and make it public:
https://github.com/gazebosim/gz-rendering/blob/ign-rendering6/include/ignition/rendering/base/BaseScene.hh#L802

Then the `BaseStorage::Begin` and `BaseStorage::End` iterator may help here. We replaced a few `*ByIndex` calls a while back for perf improvement, e.g. 
https://github.com/gazebosim/gz-rendering/blob/ign-rendering6/include/ignition/rendering/base/BaseVisual.hh#L276

darksylinc, 2022-06-06:
Yes, that would fix the problem (also documentation should mention about this behavior in SensorByIndex).

And if exposing the class is too cumbersome, it can be hidden away by a Class that provides iterators but hides the underneath storage class.

iche033, 2024-01-23:
I think we can close this as the BaseStorage class was updated to use `std::vector` instead of `std::map` in #831

closed 2024-01-23 by iche033