gazebosim/gz-rendering #683: Run tests for available render engines

iche033, 2022-07-21:
Currently we only run tests for engines specified in the RENDER_ENGINE_VALUES variable. If an engine is not available, it just skips the test. Additionally, `ogre` and `ogre2` tests can not be run one after another due to possible symbol collision. It would be nice to have a way to run tests for all render engine that are available.

## Desired behavior
Run tests for all render engine that are available.

## Alternatives considered
Run test one after another but 

## Implementation suggestion
One suggestion is to generate separate test executables for each render engine, e.g. `UNIT_ogre_Scene_TEST` and `UNIT_ogre2_Scene_TEST`

## Additional context
Related comment: https://github.com/gazebosim/gz-rendering/pull/670#issuecomment-1191004692
Related issue: https://github.com/gazebosim/gz-rendering/issues/455

mjcarroll, 2022-07-25:
> One suggestion is to generate separate test executables for each render engine, e.g. UNIT_ogre_Scene_TEST and UNIT_ogre2_Scene_TEST

I like this idea better than the current mechanism +1

iche033, 2023-10-06:
#685 and #694

closed 2023-10-06 by iche033