gazebosim/gz-rendering #695: Ogre2 Tests Failing in Metal transition

mjcarroll, 2022-07-29:
## Environment
* OS Version: macOS
* Rendering Plugin: ogre2
* Rendering Backend: Metal

## Description

When transitioning to metal, three new test failures were introduced.  They are captured here so that we remember to address them.

- [ ] UNIT_Utils_TEST `UtilTest.ClickToScene`

<details>
```
[----------] 1 test from UtilTest
185: [ RUN      ] UtilTest.ClickToScene
185: [Wrn] [Ogre2RenderTarget.cc:585] Anti-aliasing level of '0' is not supported; valid FSAA levels are: [ 1 2 4 8 ]. Setting to 1
185: [Dbg] [CommonRenderingTest.hh:76] Using OGRE2-METAL backend to test
185: [Msg] Loading plugin [gz-rendering-ogre2]
185: /Users/jenkins/mjcarroll/gz_garden/src/gz-rendering/test/common_test/Utils_TEST.cc:123: Failure
185: The difference between 0.0 and result.X() is 0.026159780099987984, which exceeds 2e-6, where
185: 0.0 evaluates to 0,
185: result.X() evaluates to 0.026159780099987984, and
185: 2e-6 evaluates to 1.9999999999999999e-06.
185: /Users/jenkins/mjcarroll/gz_garden/src/gz-rendering/test/common_test/Utils_TEST.cc:124: Failure
185: The difference between 0.0 and result.Y() is 0.026159882545471191, which exceeds 2e-6, where
185: 0.0 evaluates to 0,
185: result.Y() evaluates to 0.026159882545471191, and
185: 2e-6 evaluates to 1.9999999999999999e-06.
185: /Users/jenkins/mjcarroll/gz_garden/src/gz-rendering/test/common_test/Utils_TEST.cc:132: Failure
185: The difference between 0.0 and result.X() is 0.026159780099987984, which exceeds 2e-6, where
185: 0.0 evaluates to 0,
185: result.X() evaluates to 0.026159780099987984, and
185: 2e-6 evaluates to 1.9999999999999999e-06.
185: /Users/jenkins/mjcarroll/gz_garden/src/gz-rendering/test/common_test/Utils_TEST.cc:133: Failure
185: The difference between 0.0 and result.Y() is 0.026159882545471191, which exceeds 2e-6, where
185: 0.0 evaluates to 0,
185: result.Y() evaluates to 0.026159882545471191, and
185: 2e-6 evaluates to 1.9999999999999999e-06.
185: /Users/jenkins/mjcarroll/gz_garden/src/gz-rendering/test/common_test/Utils_TEST.cc:145: Failure
185: The difference between 0.0 and result.X() is 0.011728753335773945, which exceeds 2e-6, where
185: 0.0 evaluates to 0,
185: result.X() evaluates to 0.011728753335773945, and
185: 2e-6 evaluates to 1.9999999999999999e-06.
185: /Users/jenkins/mjcarroll/gz_garden/src/gz-rendering/test/common_test/Utils_TEST.cc:146: Failure
185: The difference between 0.0 and result.Y() is 0.011728801764547825, which exceeds 2e-6, where
185: 0.0 evaluates to 0,
185: result.Y() evaluates to 0.011728801764547825, and
185: 2e-6 evaluates to 1.9999999999999999e-06.
185: [  FAILED  ] UtilTest.ClickToScene (189 ms)

```
</details>

* [ ] INTEGRATION_Camera  `CameraTest.VisualAt`

<details>
```
[ RUN      ] CameraTest.VisualAt
9: [Dbg] [CommonRenderingTest.hh:76] Using OGRE2-METAL backend to test
9: [Msg] Loading plugin [gz-rendering-ogre2]
9: /Users/jenkins/mjcarroll/gz_garden/src/gz-rendering/test/integration/camera.cc:223: Failure
9: Expected: (nullptr) != (vis), actual: (nullptr) vs (nullptr)
9: X: 200
9: /Users/jenkins/mjcarroll/gz_garden/src/gz-rendering/test/integration/camera.cc:223: Failure
9: Expected: (nullptr) != (vis), actual: (nullptr) vs (nullptr)
9: X: 300
9: /Users/jenkins/mjcarroll/gz_garden/src/gz-rendering/test/integration/camera.cc:237: Failure
9: Expected: (nullptr) != (vis), actual: (nullptr) vs (nullptr)
9: X: 500
9: /Users/jenkins/mjcarroll/gz_garden/src/gz-rendering/test/integration/camera.cc:237: Failure
[... 111 lines cut ...]
25:   maxVal
25:     Which is: inf
25:   ry
25:     Which is: 1
25: /Users/jenkins/mjcarroll/gz_garden/src/gz-rendering/test/integration/render_pass.cc:295: Failure
25: Expected equality of these values:
25:   maxVal
25:     Which is: inf
25:   rz
25:     Which is: 1
25: /Users/jenkins/mjcarroll/gz_garden/src/gz-rendering/test/integration/render_pass.cc:315: Failure
25: Expected: (mb) > (0u), actual: 0 vs 0
25: /Users/jenkins/mjcarroll/gz_garden/src/gz-rendering/test/integration/render_pass.cc:325: Failure
25: The difference between 255u and lr is 255, which exceeds colorNoiseTol, where
25: 255u evaluates to 255,
25: lr evaluates to 0, and
25: colorNoiseTol evaluates to 10.
25: /Users/jenkins/mjcarroll/gz_garden/src/gz-rendering/test/integration/render_pass.cc:336: Failure
25: The difference between 255u and rr is 255, which exceeds colorNoiseTol, where
25: 255u evaluates to 255,
25: rr evaluates to 0, and
25: colorNoiseTol evaluates to 10.
25: [  FAILED  ] RenderPassTest.DepthGaussianNoise (259 ms)
```
</details>

mjcarroll, 2022-07-29:
CC: @darksylinc

darksylinc, 2022-07-29:
>  Metal SL Compiler Error in _gz_simple_color_330_vs.glsl:

I got confused by this error until I realized it says METAL but the shader is GLSL.

The test is simply faulty because it does not provide a Metal version for Gazebo to use. It should be easy to port and I think we have a very similar shader already ported (in gz-rendering/ogre2/src/media/materials/programs/Metal/plain_color_fs.metal).

mjcarroll, 2022-07-29:
> I got confused by this error until I realized it says METAL but the shader is GLSL.

Yeah, sorry I traced this one down a few minutes ago and came to the same conclusion.  It's a shader that is in `test/media` and not `src/media`.

darksylinc, 2022-07-29:
It's really late and I think this shader should be the ported version (untested code):

```metal
/*
 * Copyright (C) 2022 Open Source Robotics Foundation
 *
 * Licensed under the Apache License, Version 2.0 (the "License");
 * you may not use this file except in compliance with the License.
 * You may obtain a copy of the License at
 *
 *     http://www.apache.org/licenses/LICENSE-2.0
 *
 * Unless required by applicable law or agreed to in writing, software
 * distributed under the License is distributed on an "AS IS" BASIS,
 * WITHOUT WARRANTIES OR CONDITIONS OF ANY KIND, either express or implied.
 * See the License for the specific language governing permissions and
 * limitations under the License.
 *
 */
 
#include <metal_stdlib>
using namespace metal;

struct VS_INPUT
{
  float4 position [[attribute(VES_POSITION)]];
};

struct PS_INPUT
{
  float4 gl_Position [[position]];
};

struct Params
{
  float4x4 worldviewproj_matrix;
};

vertex PS_INPUT main_metal
(
  VS_INPUT input [[stage_in]],
  constant Params &p [[buffer(PARAMETER_SLOT)]]
)
{
  PS_INPUT outVs;

  outVs.gl_Position = ( p.worldViewProj * input.position ).xyzw;

  return outVs;
}
```

```metal
/*
 * Copyright (C) 2022 Open Source Robotics Foundation
 *
 * Licensed under the Apache License, Version 2.0 (the "License");
 * you may not use this file except in compliance with the License.
 * You may obtain a copy of the License at
 *
 *     http://www.apache.org/licenses/LICENSE-2.0
 *
 * Unless required by applicable law or agreed to in writing, software
 * distributed under the License is distributed on an "AS IS" BASIS,
 * WITHOUT WARRANTIES OR CONDITIONS OF ANY KIND, either express or implied.
 * See the License for the specific language governing permissions and
 * limitations under the License.
 *
 */
 
#include <metal_stdlib>
using namespace metal;

struct PS_INPUT
{
};

fragment float4 main_metal
(
  PS_INPUT inPs [[stage_in]]
)
{
  return float4(1.0, 0.0, 0.0, 1.0);
}
```