gazebosim/gz-rendering #710: Ogre crash when moving gui camera away from main scene

jennuine, 2022-08-22:
## Environment
* OS Version: Ubuntu 20.04 & 18.04
* Source or binary build?
  both

## Description
* Expected behavior: Be able to move user/gui camera off main scene w/o it crashing
* Actual behavior: The crash happens when the gui camera is moved off the main scene. This was first detected in https://github.com/gazebosim/gz-sim/pull/1499, when the HFOV is set to high and the camera moved off the main scene, the crash happens immediately and reliably. It also happens without the PR but not as reliably and can take some time for it to occur (see  https://github.com/gazebosim/gz-sim/pull/1499#pullrequestreview-1009122496 for more details).

Crash backtrace:
```bash
libEGL warning: DRI2: failed to create dri screen
libEGL warning: Not allowed to force software rendering when API explicitly selects a hardware device.
libEGL warning: DRI2: failed to create dri screen
ign gazebo gui: /var/lib/jenkins/workspace/ogre-2.2-debbuilder/repo/OgreMain/include/OgreAxisAlignedBox.h:251: void Ogre::AxisAlignedBox::setExtents(const Ogre::Vector3&, const Ogre::Vector3&): Assertion (min.x <= max.x && min.y <= max.y && min.z <= max.z) && "The minimum corner of the box must be less than or equal to maximum corner"' failed.
Stack trace (most recent call last) in thread 5410:
#31   Object "/usr/lib/x86_64-linux-gnu/libQt5Core.so.5", at 0x7fd8117278d9, in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>)
#30   Object "/usr/lib/x86_64-linux-gnu/libQt5Core.so.5", at 0x7fd81178285e, in QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>)
#29   Object "/usr/lib/x86_64-linux-gnu/libglib-2.0.so.0", at 0x7fd80d0d37fb, in g_main_context_iteration
#28   Object "/usr/lib/x86_64-linux-gnu/libglib-2.0.so.0", at 0x7fd80d0d376f, in 
#27   Object "/usr/lib/x86_64-linux-gnu/libglib-2.0.so.0", at 0x7fd80d0d3536, in g_main_context_dispatch
#26   Object "/usr/lib/x86_64-linux-gnu/libQt5Core.so.5", at 0x7fd811783232, in 
#25   Object "/usr/lib/x86_64-linux-gnu/libQt5Core.so.5", at 0x7fd81172c01c, in QCoreApplicationPrivate::sendPostedEvents(QObject*, int, QThreadData*)
#24   Object "/usr/lib/x86_64-linux-gnu/libQt5Core.so.5", at 0x7fd8117298a7, in QCoreApplication::notifyInternal2(QObject*, QEvent*)
#23   Object "/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5", at 0x7fd810b14103, in QApplication::notify(QObject*, QEvent*)
#22   Object "/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5", at 0x7fd810b0c83b, in QApplicationPrivate::notify_helper(QObject*, QEvent*)
#21   Object "/usr/lib/x86_64-linux-gnu/libQt5Core.so.5", at 0x7fd811759091, in QObject::event(QEvent*)
#20   Object "/home/tuila/Workspaces/ignFortress/install/lib/ign-gui-6/plugins/libMinimalScene.so", at 0x7fd7cbb86b4f, in ignition::gui::plugins::RenderThread::RenderNext(ignition::gui::plugins::RenderSync*)
#19   Object "/home/tuila/Workspaces/ignFortress/install/lib/ign-gui-6/plugins/libMinimalScene.so", at 0x7fd7cbb8698f, in ignition::gui::plugins::IgnRenderer::Render(ignition::gui::plugins::RenderSync*)
#18   Object "/home/tuila/Workspaces/ignFortress/install/lib/ign-rendering-6/engine-plugins/libignition-rendering-ogre2.so", at 0x7fd7a25c9242, in ignition::rendering::v6::BaseCamera<ignition::rendering::v6::Ogre2Sensor>::Update()
#17   Object "/home/tuila/Workspaces/ignFortress/install/lib/ign-rendering-6/engine-plugins/libignition-rendering-ogre2.so", at 0x7fd7a263f0c1, in ignition::rendering::v6::Ogre2RenderTarget::Render()
#16   Object "/usr/lib/x86_64-linux-gnu/libOgreMain.so.2.2.6", at 0x7fd7a204fdf2, in Ogre::CompositorWorkspace::_update(bool)
#15   Object "/usr/lib/x86_64-linux-gnu/libOgreMain.so.2.2.6", at 0x7fd7a204fb21, in Ogre::CompositorWorkspace::_update(bool)
#14   Object "/usr/lib/x86_64-linux-gnu/libOgreMain.so.2.2.6", at 0x7fd7a203eb35, in Ogre::CompositorNode::_update(Ogre::Camera const*, Ogre::SceneManager*)
#13   Object "/usr/lib/x86_64-linux-gnu/libOgreMain.so.2.2.6", at 0x7fd7a20649bc, in Ogre::CompositorPassScene::execute(Ogre::Camera const*)
#12   Object "/usr/lib/x86_64-linux-gnu/libOgreMain.so.2.2.6", at 0x7fd7a20491d6, in Ogre::CompositorShadowNode::_update(Ogre::Camera*, Ogre::Camera const*, Ogre::SceneManager*)
#11   Object "/usr/lib/x86_64-linux-gnu/libOgreMain.so.2.2.6", at 0x7fd7a203eb35, in Ogre::CompositorNode::_update(Ogre::Camera const*, Ogre::SceneManager*)
#10   Object "/usr/lib/x86_64-linux-gnu/libOgreMain.so.2.2.6", at 0x7fd7a2064ac8, in Ogre::CompositorPassScene::execute(Ogre::Camera const*)
#9    Object "/usr/lib/x86_64-linux-gnu/libOgreMain.so.2.2.6", at 0x7fd7a1d3e27e, in Ogre::Camera::_cullScenePhase01(Ogre::Camera*, Ogre::Camera const*, Ogre::Viewport*, unsigned char, unsigned char, bool)
#8    Object "/usr/lib/x86_64-linux-gnu/libOgreMain.so.2.2.6", at 0x7fd7a1f27d38, in Ogre::SceneManager::_cullPhase01(Ogre::Camera*, Ogre::Camera*, Ogre::Camera const*, unsigned char, unsigned char, bool)
#7    Object "/usr/lib/x86_64-linux-gnu/libOgreMain.so.2.2.6", at 0x7fd7a1ee4d89, in Ogre::RenderQueue::renderPassPrepare(bool, bool)
#6    Object "/usr/lib/x86_64-linux-gnu/libOgreHlmsPbs.so.2.2.6", at 0x7fd7a198b217, in Ogre::HlmsPbs::preparePassHash(Ogre::CompositorShadowNode const*, bool, bool, Ogre::SceneManager*)
#5    Object "/usr/lib/x86_64-linux-gnu/libOgreMain.so.2.2.6", at 0x7fd7a1d82669, in Ogre::Frustum::getProjectionMatrixWithRSDepth() const
#4    Object "/usr/lib/x86_64-linux-gnu/libOgreMain.so.2.2.6", at 0x7fd7a1d835fe, in Ogre::Frustum::updateFrustumImpl() const
#3    Object "/lib/x86_64-linux-gnu/libc.so.6", at 0x7fd817fa6471, in __assert_fail
#2    Object "/lib/x86_64-linux-gnu/libc.so.6", at 0x7fd817fa63f9, in 
#1    Object "/lib/x86_64-linux-gnu/libc.so.6", at 0x7fd817fb67f0, in abort
#0    Object "/lib/x86_64-linux-gnu/libc.so.6", at 0x7fd817fb4e87, in gsignal
```

## Steps to reproduce

If testing with https://github.com/gazebosim/gz-sim/pull/1499 (see below gif):
1. `ign gazebo -v 4 minimal_scene.sdf` 
2. Expand the `View Angle` plugin & update `Horizontal FOV` to `150`
3. Move user camera off main scene

To test w/o PR:
1. Add `<horizontal_fov>150</horizontal_fov>` as a child of the [`<plugin>` here
](https://github.com/gazebosim/gz-sim/blob/827fc3f8169c2e7388c66d0a41a14f10fe6c689e/examples/worlds/minimal_scene.sdf#L30)
2. `ign gazebo -v 4 /path/to/minimal_scene.sdf`
3. Move user camera off main scene

## Output

[image]

iche033, 2022-11-12:
Here is another backtrace with showing more ogre calls:

<details><summary>backtrace</summary>

```bash
(gdb) bt
#0  __GI_raise (sig=sig@entry=6) at ../sysdeps/unix/sysv/linux/raise.c:50
#1  0x00007ffff7a61859 in __GI_abort () at abort.c:79
#2  0x00007ffff7a61729 in __assert_fail_base
    (fmt=0x7ffff7bf7588 "%s%s%s:%u: %s%sAssertion `%s' failed.\n%n", assertion=0x7fffa9801200 "(min.x <= max.x && min.y <= max.y && min.z <= max.z) && \"The minimum corner of the box must be less than or equal to maximum corner\"", file=0x7fffa98011a0 "/var/lib/jenkins/workspace/ogre-2.2-debbuilder/repo/OgreMain/include/OgreAxisAlignedBox.h", line=250, function=<optimized out>) at assert.c:92
#3  0x00007ffff7a72f36 in __GI___assert_fail
    (assertion=0x7fffa9801200 "(min.x <= max.x && min.y <= max.y && min.z <= max.z) && \"The minimum corner of the box must be less than or equal to maximum corner\"", file=0x7fffa98011a0 "/var/lib/jenkins/workspace/ogre-2.2-debbuilder/repo/OgreMain/include/OgreAxisAlignedBox.h", line=250, function=0x7fffa9801148 "void Ogre::AxisAlignedBox::setExtents(const Ogre::Vector3&, const Ogre::Vector3&)") at assert.c:101
#4  0x00007fffa94f373f in Ogre::Frustum::updateFrustumImpl() const () at /usr/lib/x86_64-linux-gnu/libOgreMain.so.2.2.6
#5  0x00007fffa94f2a1e in Ogre::Frustum::getProjectionMatrixWithRSDepth() const () at /usr/lib/x86_64-linux-gnu/libOgreMain.so.2.2.6
#6  0x00007fffa92eb4a6 in Ogre::HlmsPbs::preparePassHash(Ogre::CompositorShadowNode const*, bool, bool, Ogre::SceneManager*) ()
    at /usr/lib/x86_64-linux-gnu/libOgreHlmsPbs.so.2.2.6
#7  0x00007fffa965301c in Ogre::RenderQueue::renderPassPrepare(bool, bool) () at /usr/lib/x86_64-linux-gnu/libOgreMain.so.2.2.6
#8  0x00007fffa9698ca5 in Ogre::SceneManager::_cullPhase01(Ogre::Camera*, Ogre::Camera*, Ogre::Camera const*, unsigned char, unsigned char, bool) ()
    at /usr/lib/x86_64-linux-gnu/libOgreMain.so.2.2.6
#9  0x00007fffa94af02e in Ogre::Camera::_cullScenePhase01(Ogre::Camera*, Ogre::Camera const*, Ogre::Viewport*, unsigned char, unsigned char, bool) ()
    at /usr/lib/x86_64-linux-gnu/libOgreMain.so.2.2.6
#10 0x00007fffa97dea69 in Ogre::CompositorPassScene::execute(Ogre::Camera const*) () at /usr/lib/x86_64-linux-gnu/libOgreMain.so.2.2.6
#11 0x00007fffa97b2bb5 in Ogre::CompositorNode::_update(Ogre::Camera const*, Ogre::SceneManager*) () at /usr/lib/x86_64-linux-gnu/libOgreMain.so.2.2.6
#12 0x00007fffa97be564 in Ogre::CompositorShadowNode::_update(Ogre::Camera*, Ogre::Camera const*, Ogre::SceneManager*) ()
    at /usr/lib/x86_64-linux-gnu/libOgreMain.so.2.2.6
#13 0x00007fffa97de960 in Ogre::CompositorPassScene::execute(Ogre::Camera const*) () at /usr/lib/x86_64-linux-gnu/libOgreMain.so.2.2.6
#14 0x00007fffa97b2bb5 in Ogre::CompositorNode::_update(Ogre::Camera const*, Ogre::SceneManager*) () at /usr/lib/x86_64-linux-gnu/libOgreMain.so.2.2.6
#15 0x00007fffa97c5528 in Ogre::CompositorWorkspace::_update(bool) () at /usr/lib/x86_64-linux-gnu/libOgreMain.so.2.2.6
#16 0x00007fffa97c57fb in Ogre::CompositorWorkspace::_update(bool) () at /usr/lib/x86_64-linux-gnu/libOgreMain.so.2.2.6
#17 0x00007fffa9c15da6 in ignition::rendering::v6::Ogre2RenderTarget::Render() (this=0x7fff9cfb28a8)
    at /home/ian/code/ign_f_ws/src/ign-rendering/ogre2/src/Ogre2RenderTarget.cc:474
#18 0x00007fffa9b8b6ee in ignition::rendering::v6::BaseCamera<ignition::rendering::v6::Ogre2Sensor>::Update() (this=0x7fff9ce87af0)
    at /usr/include/c++/9/ext/atomicity.h:82
#19 0x00007fffc1087de4 in ignition::gui::plugins::IgnRenderer::Render(ignition::gui::plugins::RenderSync*) (this=0x555556f94ff0, _renderSync=0x555556f94e78)
    at /usr/include/c++/9/bits/shared_ptr_base.h:1020
#20 0x00007fffc1088046 in ignition::gui::plugins::RenderThread::RenderNext(ignition::gui::plugins::RenderSync*) (this=0x555556f94fb0, _renderSync=0x555556f94e78)
    at /home/ian/code/ign_f_ws/src/ign-gui/src/plugins/minimal_scene/MinimalScene.cc:690
#21 0x00007ffff26b6c2a in QObject::event(QEvent*) () at /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#22 0x00007ffff1c3ba66 in QApplicationPrivate::notify_helper(QObject*, QEvent*) () at /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5
#23 0x00007ffff1c450f0 in QApplication::notify(QObject*, QEvent*) () at /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5
#24 0x00007ffff268a80a in QCoreApplication::notifyInternal2(QObject*, QEvent*) () at /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#25 0x00007ffff268d488 in QCoreApplicationPrivate::sendPostedEvents(QObject*, int, QThreadData*) () at /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#26 0x00007ffff26e2e37 in  () at /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#27 0x00007ffff02c117d in g_main_context_dispatch () at /usr/lib/x86_64-linux-gnu/libglib-2.0.so.0
#28 0x00007ffff02c1400 in  () at /usr/lib/x86_64-linux-gnu/libglib-2.0.so.0
#29 0x00007ffff02c14a3 in g_main_context_iteration () at /usr/lib/x86_64-linux-gnu/libglib-2.0.so.0
#30 0x00007ffff26e2435 in QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) () at /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#31 0x00007ffff26893ab in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) () at /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#32 0x00007ffff24c1785 in QThread::exec() () at /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#33 0x00007ffff24c29d2 in  () at /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#34 0x00007ffff7a22609 in start_thread (arg=<optimized out>) at pthread_create.c:477
#35 0x00007ffff7b5e293 in clone () at ../sysdeps/unix/sysv/linux/x86_64/clone.S:95
```
</details>

iche033, 2022-11-29:
after some debugging, turns out that the large hfov is not the root cause of the crash but it indirectly makes this crash easier to occur.  A GUI camera with large FOV means mouse drag movements computes to larger camera translations. Eventually, the camera pos is set to a really large world pos which likely results in `nan` / `inf` values somewhere in the maths for shadow node frustum calculation in ogre. Note the crash only happens if there is a shadow casting directional light in the scene - which is almost always the case in gazebo.

To easily reproduce this issue, just launch `minimal_scene.sdf`, go to `View Angle` gui plugin and set it's x pos to a large number, e.g. `100000000000`.

To be fair, I don't think ogre ever expects the user camera to be positioned at such a large offset from the origin. We've also seen weird rendering issues when objects are located far away from the origin so we should advice users not to position the objects this far away.  

To prevent this crash, the quick and dirty workaround would be to set a limit on the max distance the camera is allowed from the origin.

iche033, 2023-02-24:
added working around in https://github.com/gazebosim/gz-rendering/pull/824

closed 2023-02-27 by iche033