gazebosim/gz-rendering #712: GPU lidar sensor misses heightmap visuals

hidmic, 2022-08-29:
## Environment
* OS Version: Ubuntu 20.04
* Source build, current [`gz-garden`](https://github.com/gazebo-tooling/gazebodistro/blob/master/collection-garden.yaml)
* If this is a GUI or sensor rendering bug, describe your GPU and rendering system. Otherwise delete this section. 
    - Rendering plugin: ogre2. 
      - No relevant logs 
    - Generally, mention all circumstances that might affect rendering capabilities: 
      - [ ] running on a dual GPU machine (integrated GPU + discrete GPU)
      - [ ] running on a multi-GPU machine (it has multiple discrete GPUs)
      - [x] running on real hardware
      - [ ] running in virtual machine
      - [x] running in Docker/Singularity
      - [ ] running remotely (e.g. via SSH)
      - [ ] running in a cloud
      - [ ] using VirtualGL, XVFB, Xdummy, XVNC or other indirect rendering utilities
      - [ ] GPU is concurrently used for other tasks
        - [ ] desktop acceleration
        - [ ] video decoding (i.e. a playing Youtube video)
        - [ ] video encoding
        - [ ] CUDA/ROCm computations (Tensorflow, Torch, Caffe running)
        - [ ] multiple simulators running at the same time
      - [ ] other...
    - Rendering system info:
      - On Linux, provide the outputs of the following commands:
          ```bash
          $ LANG=C lspci -nn | grep VGA  # might require installing pciutils
          00:02.0 VGA compatible controller [0300]: Intel Corporation UHD Graphics 630 (Mobile) [8086:3e9b]
          01:00.0 VGA compatible controller [0300]: NVIDIA Corporation GP107M [GeForce GTX 1050 Ti Mobile] [10de:1c8c] (rev a1)          
          $ echo "$DISPLAY"
          :1
          $ LANG=C glxinfo -B | grep -i '\(direct rendering\|opengl\|profile\)'  # might require installing mesa-utils package
          direct rendering: Yes
          OpenGL vendor string: NVIDIA Corporation
          OpenGL renderer string: NVIDIA GeForce GTX 1050 Ti with Max-Q Design/PCIe/SSE2
          OpenGL core profile version string: 4.6.0 NVIDIA 470.141.03
          OpenGL core profile shading language version string: 4.60 NVIDIA
          OpenGL core profile context flags: (none)
          OpenGL core profile profile mask: core profile
          OpenGL version string: 4.6.0 NVIDIA 470.141.03
          OpenGL shading language version string: 4.60 NVIDIA
          OpenGL context flags: (none)
          OpenGL profile mask: (none)
          OpenGL ES profile version string: OpenGL ES 3.2 NVIDIA 470.141.03
          OpenGL ES profile shading language version string: OpenGL ES GLSL ES 3.20
          $ ps aux | grep Xorg  # outside container
          root        1279  0.0  0.3 25609504 119844 tty1  Sl+  08:28   0:03 /usr/lib/xorg/Xorg vt1 -displayfd 3 -auth /run/user/121/gdm/Xauthority -background none -noreset -keeptty -verbose 3
          root        2275  2.9  0.5 25670656 178144 tty2  Sl+  08:30   3:41 /usr/lib/xorg/Xorg vt2 -displayfd 3 -auth /run/user/1000/gdm/Xauthority -background none -noreset -keeptty -verbose 3
          $ sudo env LANG=C X -version  # if you don't have root access, try to tell the version of Xorg e.g. via package manager
          X.Org X Server 1.20.13
          X Protocol Version 11, Revision 0
          Build Operating System: linux Ubuntu
          Current Operating System: Linux gcrino 5.4.0-125-generic #141-Ubuntu SMP Wed Aug 10 13:42:03 UTC 2022 x86_64
          Kernel command line: BOOT_IMAGE=/boot/vmlinuz-5.4.0-125-generic root=UUID=e04e4125-9aed-4288-a2f3-27faab769cad ro quiet splash vt.handoff=7
          Build Date: 06 July 2022  01:53:24PM
          xorg-server 2:1.20.13-1ubuntu1~20.04.3 (For technical support please see http://www.ubuntu.com/support) 
          Current version of pixman: 0.38.4
	      Before reporting problems, check http://wiki.x.org
              to make sure that you have the latest version.
          ```
    - [x] Please, attach the ogre.log or [ogre2.log](https://github.com/gazebosim/gz-rendering/files/9445079/ogre2.log)
 file from  `~/.gz/rendering` 
  

## Description
* Expected behavior: GPU lidar sensor can see heightmap textures
* Actual behavior: GPU lidar sensor cannot see heightmap textures

## Steps to reproduce

1. Run `gz sim` [`repro.sdf`](https://github.com/gazebosim/gz-rendering/files/9445087/repro.zip)

3. Use PointCloud GUI plugin to display scan points.

## Output

Lidar can see the box but no the terrain (note the pointcloud position is off because of assumptions in the GUI plugin).

[image]

hidmic, 2022-08-29:
This used to be working not that long ago. I cannot confirm but I suspect #553 may have had something to do with this.

iche033, 2022-08-31:
I think visibility mask broke this. Before this PR https://github.com/gazebosim/gz-rendering/pull/625, the value is set to `IGN_VISIBILITY_ALL` which is `0x0FFFFFFF`. Now it grabs the [default value](https://github.com/gazebosim/sdformat/blob/sdf12/src/Lidar.cc#L65) from sdf which is `0xFFFFFFFF`.

One workaround is to add a `<visibility_mask>` to a different value then the lidar should start detecting the heightmap again.

hidmic, 2022-08-31:
@iche033 ahh, good catch! That was it. Is this documented somewhere?

iche033, 2022-08-31:
you mean what the visibility masks / flag ares? We added it to the [sdf spec](https://github.com/gazebosim/sdformat/blob/sdf12/sdf/1.9/lidar.sdf#L74-L76) but looks like the sdformat website hasn't been updated to pick up the news changes

azeey, 2022-10-17:
Other than the update to the sdformat docs (which I'll try to do today), anything else left to do on this issue?

iche033, 2022-11-07:
yeah we still need to fix this issue in gz-rendering. I think one potential workaround is to change [this line](https://github.com/gazebosim/gz-rendering/blob/d32ecb84233748efcebbe5ba8e965df835f40e54/ogre2/src/Ogre2GpuRays.cc#L964) from 

```
this->VisibilityMask()
```

to 

```
(this->VisibilityMask() & IGN_VISIBILITY_ALL)
```

and that should fix heightmap detection without explicitly setting the visibility mask. Oh and also add an integration test.

closed 2022-11-15 by iche033