gazebosim/gz-rendering #729: Support wide-angle cameras in ogre2

scpeters, 2022-09-22:
## Desired behavior

Support for wide-angle cameras was ported from gazebo-classic to gz-rendering's `ogre` (1.x) component in https://github.com/gazebosim/gz-rendering/pull/490. This ticket requests that it also be ported to the `ogre2` (2.x) component.

## Alternatives considered

## Implementation suggestion

## Additional context

darksylinc, 2022-10-08:
As I'm porting this to OgreWideAngleCamera to Ogre2; I noticed `OgreWideAngleCamera::Implementation::envCubeMapTextureFormat` **doesn't appear to be ever initialized**.

That looks like a bug?

scpeters, 2022-10-10:
> As I'm porting this to OgreWideAngleCamera to Ogre2; I noticed `OgreWideAngleCamera::Implementation::envCubeMapTextureFormat` **doesn't appear to be ever initialized**.
> 
> That looks like a bug?
> 
> @iche033

that looks like a bug to me

iche033, 2022-10-10:
> that looks like a bug to me
 
+1.

darksylinc, 2022-12-03:
This ticket can be closed. PR #733 implemented this feature.

PR #733 (2022-10-31): Add Support for wide-angle cameras in ogre2
# 🎉 New feature

Closes #729 

## Summary

This PR ports OgreWideAngleCamera to Ogre2 engine.

Includes Vulkan, OpenGL & Metal shaders (Metal support not tested)

## Test it

Edit `wide_angle_camera.sdf` so that:

```
    <plugin
      filename="gz-sim-sensors-system"
      name="gz::sim::systems::Sensors">
      <render_engine>ogre</render_engine>
    </plugin>
```

No longer says ogre and run:

`gz sim wide_angle_camera.sdf`

## Can it be backported to Garden?

The only major/blocking issue I see is that `Ogre2Scene::CreateWideAngleCameraImpl` was  [... cut]
closed 2022-12-05 by iche033