gazebosim/gz-rendering #731: Ubuntu Focal/arm64 build fails to compile 7.0.0

j-rivero, 2022-09-28:
# Bug report

## Environment
* OS Version: Ubuntu Focal
* Building .deb packages for 7.0.0

The build fails to link ibgz-rendering7-ogre2.so with errors comming from different Ogre elements.
The Ubuntu ARM64 Jammy build is fine, the amd64 Focal build is fine.

## Log
```
[100%] Linking CXX shared library ../../lib/libgz-rendering7-ogre2.so
cd /home/jenkins/workspace/gz-rendering7-debbuilder/build/gz-rendering-7.0.0/obj-aarch64-linux-gnu/ogre2/src && /usr/bin/cmake -E cmake_link_script CMakeFiles/gz-rendering7-ogre2.dir/link.txt --verbose=1
/usr/bin/c++ -fPIC -g -O2 -fdebug-prefix-map=/home/jenkins/workspace/gz-rendering7-debbuilder/build/gz-rendering-7.0.0=. -fstack-protector-strong -Wformat -Werror=format-security -Wdate-time -D_FORTIFY_SOURCE=2 -O2 -g -DNDEBUG  -Wall -Wextra -Wno-long-long -Wno-unused-value -Wfloat-equal -Wshadow -Winit-self -Wswitch-default -Wmissing-include-dirs -pedantic -UNDEBUG -Wl,-Bsymbolic-functions -Wl,-z,relro -shared -Wl,-soname,libgz-rendering7-ogre2.so.7 -o ../../lib/libgz-rendering7-ogre2.so.7.0.0 CMakeFiles/gz-rendering7-ogre2.dir/Ogre2ArrowVisual.cc.o CMakeFiles/gz-rendering7-ogre2.dir/Ogre2AxisVisual.cc.o CMakeFiles/gz-rendering7-ogre2.dir/Ogre2BoundingBoxCamera.cc.o CMakeFiles/gz-rendering7-ogre2.dir/Ogre2BoundingBoxMaterialSwitcher.cc.o CMakeFiles/gz-rendering7-ogre2.dir/Ogre2COMVisual.cc.o CMakeFiles/gz-rendering7-ogre2.dir/Ogre2Camera.cc.o CMakeFiles/gz-rendering7-ogre2.dir/Ogre2Capsule.cc.o CMakeFiles/gz-rendering7-ogre2.dir/Ogre2Conversions.cc.o CMakeFiles/gz-rendering7-ogre2.dir/Ogre2DepthCamera.cc.o CMakeFiles/gz-rendering7-ogre2.dir/Ogre2DynamicRenderable.cc.o CMakeFiles/gz-rendering7-ogre2.dir/Ogre2GaussianNoisePass.cc.o CMakeFiles/gz-rendering7-ogre2.dir/Ogre2Geometry.cc.o CMakeFiles/gz-rendering7-ogre2.dir/Ogre2GizmoVisual.cc.o CMakeFiles/gz-rendering7-ogre2.dir/Ogre2GpuRays.cc.o CMakeFiles/gz-rendering7-ogre2.dir/Ogre2Grid.cc.o CMakeFiles/gz-rendering7-ogre2.dir/Ogre2GzHlmsPbsPrivate.cc.o CMakeFiles/gz-rendering7-ogre2.dir/Ogre2GzHlmsSharedPrivate.cc.o CMakeFiles/gz-rendering7-ogre2.dir/Ogre2GzHlmsSphericalClipMinDistance.cc.o CMakeFiles/gz-rendering7-ogre2.dir/Ogre2GzHlmsTerraPrivate.cc.o CMakeFiles/gz-rendering7-ogre2.dir/Ogre2GzHlmsUnlitPrivate.cc.o CMakeFiles/gz-rendering7-ogre2.dir/Ogre2Heightmap.cc.o CMakeFiles/gz-rendering7-ogre2.dir/Ogre2InertiaVisual.cc.o CMakeFiles/gz-rendering7-ogre2.dir/Ogre2JointVisual.cc.o CMakeFiles/gz-rendering7-ogre2.dir/Ogre2LidarVisual.cc.o CMakeFiles/gz-rendering7-ogre2.dir/Ogre2Light.cc.o CMakeFiles/gz-rendering7-ogre2.dir/Ogre2LightVisual.cc.o CMakeFiles/gz-rendering7-ogre2.dir/Ogre2Marker.cc.o CMakeFiles/gz-rendering7-ogre2.dir/Ogre2Material.cc.o CMakeFiles/gz-rendering7-ogre2.dir/Ogre2MaterialSwitcher.cc.o CMakeFiles/gz-rendering7-ogre2.dir/Ogre2Mesh.cc.o CMakeFiles/gz-rendering7-ogre2.dir/Ogre2MeshFactory.cc.o CMakeFiles/gz-rendering7-ogre2.dir/Ogre2Node.cc.o CMakeFiles/gz-rendering7-ogre2.dir/Ogre2Object.cc.o CMakeFiles/gz-rendering7-ogre2.dir/Ogre2ObjectInterface.cc.o CMakeFiles/gz-rendering7-ogre2.dir/Ogre2ParticleEmitter.cc.o CMakeFiles/gz-rendering7-ogre2.dir/Ogre2ParticleNoiseListener.cc.o CMakeFiles/gz-rendering7-ogre2.dir/Ogre2RayQuery.cc.o CMakeFiles/gz-rendering7-ogre2.dir/Ogre2RenderEngine.cc.o CMakeFiles/gz-rendering7-ogre2.dir/Ogre2RenderPass.cc.o CMakeFiles/gz-rendering7-ogre2.dir/Ogre2RenderTarget.cc.o CMakeFiles/gz-rendering7-ogre2.dir/Ogre2RenderTargetMaterial.cc.o CMakeFiles/gz-rendering7-ogre2.dir/Ogre2Scene.cc.o CMakeFiles/gz-rendering7-ogre2.dir/Ogre2SegmentationCamera.cc.o CMakeFiles/gz-rendering7-ogre2.dir/Ogre2SegmentationMaterialSwitcher.cc.o CMakeFiles/gz-rendering7-ogre2.dir/Ogre2SelectionBuffer.cc.o CMakeFiles/gz-rendering7-ogre2.dir/Ogre2Sensor.cc.o CMakeFiles/gz-rendering7-ogre2.dir/Ogre2ThermalCamera.cc.o CMakeFiles/gz-rendering7-ogre2.dir/Ogre2Visual.cc.o CMakeFiles/gz-rendering7-ogre2.dir/Ogre2WireBox.cc.o   -L/usr/lib/aarch64-linux-gnu/OGRE-2.3  -Wl,-rpath,/usr/lib/aarch64-linux-gnu/OGRE-2.3:/home/jenkins/workspace/gz-rendering7-debbuilder/build/gz-rendering-7.0.0/obj-aarch64-linux-gnu/lib: ../../lib/libgz-rendering7.so.7.0.0 /usr/lib/aarch64-linux-gnu/libOpenGL.so /usr/lib/aarch64-linux-gnu/libGLX.so /usr/lib/aarch64-linux-gnu/libGLU.so ../../lib/libterra.a /usr/lib/aarch64-linux-gnu/OGRE-2.3/libOgreNextMain.so /usr/lib/aarch64-linux-gnu/libgz-common5-events.so.5.0.0 /usr/lib/aarch64-linux-gnu/libgz-common5-geospatial.so.5.0.0 /usr/lib/aarch64-linux-gnu/libgz-common5-graphics.so.5.0.0 /usr/lib/aarch64-linux-gnu/libgz-common5.so.5.0.0 /usr/lib/aarch64-linux-gnu/libuuid.so /usr/lib/aarch64-linux-gnu/libuuid.so -lpthread /usr/lib/aarch64-linux-gnu/libgz-math7.so.7.0.2 /usr/lib/aarch64-linux-gnu/libgz-plugin2.so.2.0.0 /usr/lib/aarch64-linux-gnu/libgz-utils2.so.2.0.0 /usr/lib/aarch64-linux-gnu/OGRE-2.3/libOgreNextMain.so /usr/lib/aarch64-linux-gnu/OGRE-2.3/libOgreNextHlmsPbs.so /usr/lib/aarch64-linux-gnu/OGRE-2.3/libOgreNextHlmsPbs.so /usr/lib/aarch64-linux-gnu/OGRE-2.3/libOgreNextHlmsUnlit.so /usr/lib/aarch64-linux-gnu/OGRE-2.3/libOgreNextHlmsUnlit.so /usr/lib/aarch64-linux-gnu/OGRE-2.3/libOgreNextOverlay.so /usr/lib/aarch64-linux-gnu/OGRE-2.3/libOgreNextOverlay.so /usr/lib/aarch64-linux-gnu/OGRE-2.3/libOgreNextPlanarReflections.so /usr/lib/aarch64-linux-gnu/OGRE-2.3/libOgreNextPlanarReflections.so 
/usr/bin/ld: ../../lib/libterra.a(OgreHlmsTerra.cpp.o): relocation R_AARCH64_ADR_PREL_PG_HI21 against symbol `_ZTVN4Ogre9HlmsTerraE' which may bind externally can not be used when making a shared object; recompile with -fPIC
../../lib/libterra.a(OgreHlmsTerra.cpp.o): in function `Ogre::HlmsTerra::~HlmsTerra()':
./obj-aarch64-linux-gnu/ogre2/src/terrain/Terra/./ogre2/src/terrain/Terra/src/Hlms/OgreHlmsTerra.cpp:94:(.text+0x4): dangerous relocation: unsupported relocation
/usr/bin/ld: ../../lib/libterra.a(OgreHlmsTerra.cpp.o): relocation R_AARCH64_ADR_PREL_PG_HI21 against symbol `_ZTVN4Ogre9HlmsTerraE' which may bind externally can not be used when making a shared object; recompile with -fPIC
./obj-aarch64-linux-gnu/ogre2/src/terrain/Terra/./ogre2/src/terrain/Terra/src/Hlms/OgreHlmsTerra.cpp:94:(.text+0x10): dangerous relocation: unsupported relocation
/usr/bin/ld: ../../lib/libterra.a(OgreHlmsTerra.cpp.o): relocation R_AARCH64_ADR_PREL_PG_HI21 against symbol `_ZTVN4Ogre9HlmsTerraE' which may bind externally can not be used when making a shared object; recompile with -fPIC
../../lib/libterra.a(OgreHlmsTerra.cpp.o): in function `Ogre::HlmsTerra::HlmsTerra(Ogre::Archive*, std::vector<Ogre::Archive*, Ogre::STLAllocator<Ogre::Archive*, Ogre::CategorisedAllocPolicy<(Ogre::MemoryCategory)0> > >*)':
./obj-aarch64-linux-gnu/ogre2/src/terrain/Terra/./ogre2/src/terrain/Terra/src/Hlms/OgreHlmsTerra.cpp:78:(.text+0x1b0c): dangerous relocation: unsupported relocation
/usr/bin/ld: ../../lib/libterra.a(OgreHlmsTerra.cpp.o): relocation R_AARCH64_ADR_PREL_PG_HI21 against symbol `_ZTVN4Ogre9HlmsTerraE' which may bind externally can not be used when making a shared object; recompile with -fPIC
./obj-aarch64-linux-gnu/ogre2/src/terrain/Terra/./ogre2/src/terrain/Terra/src/Hlms/OgreHlmsTerra.cpp:78:(.text+0x1b1c): dangerous relocation: unsupported relocation
/usr/bin/ld: ../../lib/libterra.a(OgreHlmsTerraDatablock.cpp.o): relocation R_AARCH64_ADR_PREL_PG_HI21 against symbol `_ZTVN4Ogre29HlmsTerraBaseTextureDatablockE' which may bind externally can not be used when making a shared object; recompile with -fPIC
../../lib/libterra.a(OgreHlmsTerraDatablock.cpp.o): in function `Ogre::HlmsTerraBaseTextureDatablock::HlmsTerraBaseTextureDatablock(Ogre::IdString, Ogre::Hlms*, Ogre::HlmsMacroblock const*, Ogre::HlmsBlendblock const*, std::vector<std::pair<Ogre::IdString, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >, Ogre::STLAllocator<std::pair<Ogre::IdString, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >, Ogre::CategorisedAllocPolicy<(Ogre::MemoryCategory)0> > > const&)':
/usr/include/OGRE-2.3/Hlms/Common/OgreHlmsTextureBaseClass.inl:26:(.text+0x8b8): dangerous relocation: unsupported relocation
/usr/bin/ld: ../../lib/libterra.a(OgreHlmsTerraDatablock.cpp.o): relocation R_AARCH64_ADR_PREL_PG_HI21 against symbol `_ZTVN4Ogre29HlmsTerraBaseTextureDatablockE' which may bind externally can not be used when making a shared object; recompile with -fPIC
/usr/include/OGRE-2.3/Hlms/Common/OgreHlmsTextureBaseClass.inl:26:(.text+0x8c8): dangerous relocation: unsupported relocation
[... cut ...]
./obj-aarch64-linux-gnu/ogre2/src/terrain/Terra/./ogre2/src/terrain/Terra/src/TerrainCell.cpp:50:(.text+0x14): dangerous relocation: unsupported relocation
/usr/bin/ld: ../../lib/libterra.a(TerrainCell.cpp.o): relocation R_AARCH64_ADR_PREL_PG_HI21 against symbol `_ZTVN4Ogre11TerrainCellE' which may bind externally can not be used when making a shared object; recompile with -fPIC
../../lib/libterra.a(TerrainCell.cpp.o): in function `Ogre::TerrainCell::TerrainCell(Ogre::Terra*)':
./obj-aarch64-linux-gnu/ogre2/src/terrain/Terra/./ogre2/src/terrain/Terra/src/TerrainCell.cpp:46:(.text+0xbc): dangerous relocation: unsupported relocation
```

iche033, 2022-09-28:
> recompile with -fPIC

Looks like we will need to make sure this flag is enabled here for arm64:
https://github.com/gazebosim/gz-rendering/blob/gz-rendering7/ogre2/src/terrain/Terra/CMakeLists.txt#L12

j-rivero, 2022-10-03:
Fixed by https://github.com/gazebo-release/gz-rendering7-release/pull/19. Thanks @iche033 !

closed 2022-10-03 by j-rivero