gazebosim/gz-rendering #745: `SEH Exception` in gz-rendering-win

Crola1702, 2022-10-18:
## Environment
* OS Version:
  * Windows
* Source build
  gz-rendering7

## Description
* Expected behavior: All test passing
* Actual behavior: 61 tests failing

## Steps to reproduce
1. Run a build on [gz-rendering7-win](https://build.osrfoundation.org/job/ign_rendering-gz-7-win/)
2. See it faili

## Output

We're 61 test regressions in gz-rendering7 Windows

Reference build: https://build.osrfoundation.org/job/ign_rendering-gz-7-win/15/

Most of them are caused by:
```
unknown file
SEH exception with code 0xc0000005 thrown in the test body
```

First time we saw it: [Sept 1](https://build.osrfoundation.org/job/ign_rendering-gz-7-win/4/)
It stopped between [Sept 28 and Oct 11](https://build.osrfoundation.org/job/ign_rendering-gz-7-win/12/)

And now it [came back](https://build.osrfoundation.org/job/ign_rendering-gz-7-win/14/)

We think it's probably related to this issue: https://stackoverflow.com/questions/13157671/seh-exception-with-code-0xc0000005-thrown-in-the-test-body

iche033, 2022-11-07:
is this still an issue?

azeey, 2023-09-27:
The SEH exception might be a red herring. Based on the error
```
11: [m[31m[Err] [C:\J\workspace\ign_rendering-gz-7-win\ws\gz-rendering\ogre\src\OgreRenderEngine.cc:695] [m[31mUnable to create the rendering window. Attempt [[m[31m0[m[31m]. Exception [[m[31mOgre::RenderingAPIException::RenderingAPIException: OpenGL 1.5 is not supported in GLRenderSystem::initialiseContext at C:\vcpkg\buildtrees\ogre\src\eddf310f0b-6ab1152694.clean\RenderSystems\GL\src\OgreGLRenderSystem.cpp (line 1167)[m[31m][m[31m
```
my guess is that the graphics card is not set up properly on that machine. The same test doesn't print that error on a passing test on a different machine (see `INTEGRATION_boundingbox_camera_ogre_gl3plus` (test 11) on https://build.osrfoundation.org/job/ign_rendering-gz-7-win/79/consoleFull#console-section-12).

mjcarroll, 2023-09-27:
Yeah, the error is probably likely that something in the scene creation process is an unchecked `nullptr`.  For the most part, you can think of the SEH `0xc0000005` as the Windows equivalent to a segfault. I believe that Windows calls it an access violation, which means that a pointer is not what it should be.

Crola1702, 2024-01-09:
This is also happening on [gz_rendering-main](https://build.osrfoundation.org/job/gz_rendering-main-win/10/).

I added Ionic as a label

Crola1702, 2024-04-08:
This is still affecting gz-rendering windows jobs with the same output

Reference builds:
* https://build.osrfoundation.org/job/gz_rendering-main-win/36/
* https://build.osrfoundation.org/job/gz_rendering-7-win/38/
* https://build.osrfoundation.org/job/gz_rendering-8-win/37/

---
Failure Percentage:

job_name|last_fail|first_fail|build_count|failure_count|failure_percentage
-- | -- | -- | -- | -- | --
gz_rendering-main-win|2024-04-08|2024-03-12|8|8|100.0
gz_rendering-7-win|2024-04-07|2024-03-10|8|8|100.0
gz_rendering-8-win|2024-04-04|2024-03-09|8|8|100.0

azeey, 2024-04-08:
@Crola1702 any progress on setting up ogre 2.3 on our windows machines? I believe that's a requirement for fixing these tests.

Crola1702, 2024-04-08:
> @Crola1702 any progress on setting up ogre 2.3 on our windows machines

I'm not sure. I can take a look with @j-rivero's help. Any ideas on where to start?

azeey, 2024-04-08:
Last time we talked about this, I remember the next action item was to check if our [vcpkg port for ogre-next-23](https://github.com/osrf/vcpkg-ports/tree/master/ogre-next-23) actually works. If you can verify that locally, I think we can make the change in https://github.com/gazebo-tooling/release-tools/blob/f392d30813b5097229b22413f56a6733556d34f4/jenkins-scripts/lib/windows_env_vars.bat#L22

iche033, 2024-08-16:
I did a quick test adding `ogre-next-23` dependency. First it failed because it tires to install files in a path that conflicts with `ogre`. I then removed `ogre`, `ogre2`, `ogre22` from the dependency list (see [ogre23](https://github.com/gazebo-tooling/release-tools/compare/master...ogre23) branch in release-tools) and installation was successful, however, gz-rendering failed to find ogre-next when running cmake. 

So I think there are a couple to tasks to do for getting ogre-next 2.3 working on windows:
* Make `ogre-next-23` side-by-side installable with `ogre`
* Either update `ogre-next-23` vcpkg to install files in a way that's consistent with `ogre22` (so that gz-rendering can find it) or update `FindGzOgre2.cmake` to find this package while making sure not to break usage on conda.

Crola1702, 2025-03-27:
I will close this issue as the conda changes solved most of them.

closed 2025-03-27 by Crola1702