gazebosim/gz-rendering #747: Does macOS still need 'legacy' Ogre2RayQuery path force-disabled?

darksylinc, 2022-10-24:
**This discussion is about Garden and/or later.**

Ogre2RayQuery uses the GPU when possible to perform ray queries.

However [currently this path is disabled for macOS](https://github.com/gazebosim/gz-rendering/blob/5686fd29525b25c714fbf1d607e4c523ab67928f/ogre2/src/Ogre2RayQuery.cc#L134-L156):

```cpp
#ifdef __APPLE__
  return this->ClosestPointByIntersection(_forceSceneUpdate);
#else
   ...
#endif
```

I strongly suspect the cause was either that OpenGL3+ path was malfunctioning on macOS, or that we didn't have the proper Metal shaders for SelectionCamera (which we should now, because the code was moved from using Low Level Materials to Hlms pieces for Garden).

In other words, if I'm right, using the GPU path for Metal should "just work".

The "legacy" version is still needed; in fact the Lens Flare effect I'm working on needs to force it (I'm adding an API to set whether using the GPU is preferred).

mjcarroll, 2022-10-31:
Can we just enable it with metal in CI and see what happens?

darksylinc, 2022-10-31:
Yeah, I was also thinking we could do just that.

iche033, 2022-11-07:
> Can we just enable it with metal in CI and see what happens?

+1. It was disabled because the test was failing. I remember it was with OpenGL3+ and we didn't have proper metal support at the time. So we should just try enabling it.

scpeters, 2022-11-14:
enabled in https://github.com/gazebosim/gz-rendering/pull/758

closed 2022-11-14 by scpeters