gazebosim/gz-rendering #757: Render to external window

Levi-Armstrong, 2022-11-05:
How to render to external window? I was expecting if I set the `winID` to the QWidget window id that it would draw the content to this location. This works if I use ogre2  directly so not sure why this does not work here.

```
  std::map<std::string, std::string> params;
  params["winID"] = std::to_string(winId());
  auto* engine = ignition::rendering::engine("ogre2", params);
```

iche033, 2022-11-08:
> This works if I use ogre2 directly

Maybe we're passing different params to ogre2 when creating the render window?

the params that are passed to ogre2 when creating the render window can be found here:
https://github.com/gazebosim/gz-rendering/blob/gz-rendering7/ogre2/src/Ogre2RenderEngine.cc#L1022

you can see that we are setting the `parentWindowHandle` param to `winID` here:
https://github.com/gazebosim/gz-rendering/blob/gz-rendering7/ogre2/src/Ogre2RenderEngine.cc#L1083

Levi-Armstrong, 2022-11-08:
I did a simple demo using ogre2 directly with Qt, but setting the `parentWindowHandle` did not work either. Though if I set the `externalWindowHandle` to the `winID` it works. Do you know what the difference is between these two? I see that for different operating system it set the `externalWindowHandle` instead of the `parrentWindowHandle`

iche033, 2022-11-08:
>  Do you know what the difference is between these two?

The diff based on [ogre's doc](https://ogrecave.github.io/ogre-next/api/2.3/class_ogre_1_1_root.html#ae1c38392ce62547652f39b9a899fb20a) is that `externalWindowHandle` embeds ogre in existing window and `parentWindowHandle` embeds ogre in child of external window. I see that in the ogre wiki page on integrating Qt5, [both are set](https://wiki.ogre3d.org/Integrating+Ogre+into+QT5).

One thing to try is modify [this line](https://github.com/gazebosim/gz-rendering/blob/gz-rendering7/ogre2/src/Ogre2RenderEngine.cc#L1014) in gz-rendering to create a window of proper size and see if it attaches to the external window that you have. Some additional details: In gazebo, we're currently just creating a 1x1 window for initializing ogre. Then we do a trick in gz-gui that tells Qt window to [render the texture](https://github.com/gazebosim/gz-gui/blob/gz-gui7/src/plugins/minimal_scene/MinimalSceneRhiOpenGL.cc#L219) generated by the user camera we created in ogre.

Levi-Armstrong, 2022-11-08:
> One thing to try is modify [this line](https://github.com/gazebosim/gz-rendering/blob/gz-rendering7/ogre2/src/Ogre2RenderEngine.cc#L1014) in gz-rendering to create a window of proper size and see if [...]

This is ultimately what I want to do, but I am using QWidgets and have been unsuccessful in making this work using QOpenglWidget. Do you have or no of any examples where this has been used with QWidgets?

azeey, 2024-05-31:
Is this fixed by https://github.com/gazebosim/gz-rendering/pull/992?