gazebosim/gz-rendering #792: Ogre2 is not found on windows CI builds

azeey, 2022-12-19:
## Environment
* OS Version: Windows
* Source or binary build? source

## Description
* Expected behavior: gz-rendering to build with ogre2 in windows CI jobs
* Actual behavior: CMake is unable to find the Ogre2 package even though it's installed. See https://build.osrfoundation.org/view/ign-garden/job/ign_rendering-gz-7-win/30/consoleFull#console-section-20

<details>
<summary>Here's the output of the CMake configuration:</summary>

```
-- Selecting Windows SDK version 10.0.19041.0 to target Windows 10.0.19044.
-- The C compiler identification is MSVC 19.29.30143.0
-- The CXX compiler identification is MSVC 19.29.30143.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: C:/Program Files (x86)/Microsoft Visual Studio/2019/Community/VC/Tools/MSVC/14.29.30133/bin/Hostx64/x64/cl.exe - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: C:/Program Files (x86)/Microsoft Visual Studio/2019/Community/VC/Tools/MSVC/14.29.30133/bin/Hostx64/x64/cl.exe - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- gz-rendering7 version 7.2.0
-- Operating system is Windows
-- Looking for gz-math7 -- found version 7.1.0
-- Searching for dependencies of gz-math7
-- Looking for gz-utils2 -- found version 2.0.0
-- Searching for dependencies of gz-utils2
-- Searching for <gz-math7> component [eigen3]
-- Looking for gz-math7-eigen3 -- found version 7.1.0
-- Searching for dependencies of gz-math7-eigen3
-- Looking for gz-math7 - found

-- Looking for gz-common5 -- found version 5.3.1
-- Searching for dependencies of gz-common5
-- Looking for gz-utils2 -- found version 2.0.0
-- Found DL: TRUE  
-- Searching for <gz-common5> component [graphics]
-- Looking for gz-common5-graphics -- found version 5.3.1
-- Searching for dependencies of gz-common5-graphics
-- Looking for gz-math7 -- found version 7.1.0
-- Found GzAssimp: TRUE (found version "5.0.1") 
-- Searching for <gz-common5> component [events]
-- Looking for gz-common5-events -- found version 5.3.1
-- Searching for dependencies of gz-common5-events
-- Looking for gz-math7 -- found version 7.1.0
-- Searching for <gz-common5> component [geospatial]
-- Looking for gz-common5-geospatial -- found version 5.3.1
-- Searching for dependencies of gz-common5-geospatial
-- Looking for gz-math7 -- found version 7.1.0
-- Looking for gz-common5 -- found version 5.3.1
-- Looking for gz-common5 - found

-- Looking for gz-plugin2 -- found version 2.0.1
-- Searching for dependencies of gz-plugin2
-- Looking for gz-utils2 -- found version 2.0.0
-- Searching for dependencies of gz-utils2
-- Searching for <gz-utils2> component [cli]
-- Looking for gz-utils2-cli -- found version 2.0.0
-- Searching for dependencies of gz-utils2-cli
-- Searching for <gz-plugin2> component [all]
-- Looking for all libraries of gz-plugin2 -- found version 2.0.1
-- Looking for gz-plugin2 -- found version 2.0.1
-- Looking for gz-plugin2-loader -- found version 2.0.1
-- Searching for dependencies of gz-plugin2-loader
-- Looking for gz-plugin2-register -- found version 2.0.1
-- Searching for dependencies of gz-plugin2-register
-- Looking for gz-plugin2 - found

-- Looking for gz-utils2 -- found version 2.0.0
-- Looking for gz-utils2 - found

-- Looking for FreeImage.h FreeImageconfig.h - found
-- Looking for FreeImage library - found
-- Looking for FreeImage - found

-- Found OpenGL: opengl32   
-- Looking for OpenGL - found

-- Found OGRE
--   static     : OFF
--   components : Bites;MeshLodGenerator;Overlay;Paging;Property;RTShaderSystem;Terrain;Volume
--   plugins    : Plugin_BSPSceneManager;Plugin_OctreeSceneManager;Plugin_PCZSceneManager;Plugin_ParticleFX;RenderSystem_GL;RenderSystem_GL3Plus;RenderSystem_Direct3D11;Codec_STBI;Codec_FreeImage;Codec_EXR
--   media      : C:/vcpkg/installed/x64-windows/Media
-- Looking for GzOGRE - found

CMake Warning (dev) at C:/Jenkins/workspace/ign_rendering-gz-7-win/ws/install/gz-cmake3/share/cmake/gz-cmake3/cmake3/GzUtils.cmake:444 (message):
   -- THIS MESSAGE IS INTENDED FOR GZ-RENDERING AUTHORS --
      (IF YOU SEE THIS, PLEASE REPORT IT)

  Could not find pkg-config information for GzOGRE.  It was not provided by
  the find-module for the package, nor was it explicitly passed into the call
  to gz_find_package(~).  This is most likely an error in this project's use
  of gz-cmake.
Call Stack (most recent call first):
  CMakeLists.txt:78 (gz_find_package)
This warning is for project developers.  Use -Wno-dev to suppress it.

-- Found Boost: C:/vcpkg/installed/x64-windows/include (found version "1.78.0")  
-- Looking for GzOGRE2 - not found

CMake Warning at C:/Jenkins/workspace/ign_rendering-gz-7-win/ws/install/gz-cmake3/share/cmake/gz-cmake3/cmake3/GzConfigureBuild.cmake:69 (message):
   CONFIGURATION WARNINGS:
   -- Ogre 1.x versions greater than 1.9 are not officially supported.
   -- The software might compile and even work but support from upstream
   -- could be reduced to accepting patches for newer versions
Call Stack (most recent call first):
  CMakeLists.txt:176 (gz_configure_build)

-- Configuring library: gz-rendering7
-- Performing Test COMPILER_HAS_DEPRECATED_ATTR
-- Performing Test COMPILER_HAS_DEPRECATED_ATTR - Failed
-- Performing Test COMPILER_HAS_DEPRECATED
-- Performing Test COMPILER_HAS_DEPRECATED - Success
-- Adding 5 UNIT tests
-- Configuring library: gz-rendering7-ogre
-- No tests have been specified for UNIT
-- The program [cppcheck] was not found! Skipping codecheck setup
-- Build configuration successful
-- Build type: Release
-- Install prefix: C:/Jenkins/workspace/ign_rendering-gz-7-win/ws/install/gz-rendering7
-- Could NOT find Doxygen (missing: DOXYGEN_EXECUTABLE) 
-- Configuring done
-- Generating done
-- Build files have been written to: C:/Jenkins/workspace/ign_rendering-gz-7-win/ws/build/gz-rendering7
```
</details>

iche033, 2024-01-26:
looks like vcpkg installs ogre 2.2 but from garden (gz-rendering7) upwards, ogre 2.3 is required so the gz-rendering-ogre2 component is not built

azeey, 2024-05-31:
Closing as duplicate of https://github.com/gazebosim/gz-rendering/issues/745

closed 2024-05-31 by azeey