gazebosim/gz-rendering #810: Long camera update

Levi-Armstrong, 2023-01-22:
## Environment
* OS Version: Ubuntu 20.04 
* Source or binary build?
  * binary, rendering6

[image]

## Description
* I have noticed that after adding a geometry that I am seeing long camera updates around 30 seconds and not sure how to resolve the issue. After the geometry is added I am able to change joint state and the visualization updated without any issues but as soon as I use the mouse to try and zoom or manipulate the camera it freezes for around 30 seconds. After the long update I have no issues manipulating the camera until new objects are added to the scene again. I attached a debugger and bellow is the back trace where it is at during this long pause. I have also tried turning of shadows and turning of anti-aliasing but it has no impact on the long camera update. I am not sure where to look next or what could be the issue. Also I load the same environment in RVIZ and do not have any issues.

I have tested this with both Ogre and Ogre2 rendering and they both have the same long camera update..

[image]

Levi-Armstrong, 2023-01-22:
I have found that if this object is not in the scene it responds as I would expect. Any thoughts on what might be causing the issues? Here is a [link](https://github.com/tesseract-robotics/tesseract_ros_workcell/tree/master/twc_support/CAD/eoat/pushcorp_afd620/urdf/visual) to the dae files.

[image]

iche033, 2023-01-23:
I assume you mean the GUI camera right and not a camera sensor? I don't see anything weird with these 2 dae meshes. I just did a quick test and created a model consisting of 2 visuals, one for each dae mesh (`base_link.dae` and `link_1.dae`) but I can't seem to reproduce the long camera update problem. I did the test in `ign-rendering6` with both `ogre` and `ogre2` render engines

Levi-Armstrong, 2023-01-24:
I will try just this link and see if I see an issue.

Levi-Armstrong, 2023-01-24:
It looks like you only load the model independently they have less of an affect. Also the more models in the environment in addition to these two models increase the pause time.

- Compliance device: 3 seconds
[image]

- [Spindle](https://github.com/tesseract-robotics/tesseract_ros_workcell/blob/master/twc_support/CAD/eoat/pushcorp_stc1015/urdf/visual/base_link.dae): 6 seconds
[image]

- Compliance device + Spindle: 24 seconds

@iche033 Here is the [spindle model](https://github.com/tesseract-robotics/tesseract_ros_workcell/blob/master/twc_support/CAD/eoat/pushcorp_stc1015/urdf/visual/base_link.dae) which combined with the other model is causing the 24 second pause. It almost looks like it is loading the model again because it is around the same amount of time. Also these models are not water tight if that may have an affect.

Original Model (Compliance device)
```
Camera Update: 0.257443ms
Camera Update: 0.250413ms
Camera Update: 3405.7ms // Load model
Camera Update: 88.5545ms
Camera Update: 83.4895ms
Camera Update: 83.9671ms
Camera Update: 83.6664ms
Camera Update: 83.6583ms
Camera Update: 83.8001ms
Camera Update: 84.3852ms
Camera Update: 85.1221ms
Camera Update: 83.1592ms
Camera Update: 84.3725ms
Camera Update: 84.697ms
Camera Update: 84.8974ms
Camera Update: 84.6631ms
Camera Update: 84.5477ms
Camera Update: 3550.15ms // Click and drag
Camera Update: 95.3022ms
Camera Update: 91.2949ms
```
Just Spindle
```
Camera Update: 0.237799ms
Camera Update: 0.239147ms
Camera Update: 0.2653ms
Camera Update: 0.27567ms
Camera Update: 6331.76ms // Load Model
Camera Update: 130.058ms
Camera Update: 125.33ms
Camera Update: 126.79ms
Camera Update: 126.314ms
Camera Update: 125.869ms
Camera Update: 126.432ms
Camera Update: 127.742ms
Camera Update: 128.475ms
Camera Update: 127.231ms
Camera Update: 125.232ms
Camera Update: 6561.65ms // Click and drag
Camera Update: 140.579ms
Camera Update: 135.95ms
```

Original Model + Spindle
```
Camera Update: 0.275657ms
Camera Update: 0.24785ms
Camera Update: 22861.1ms // Load model
Camera Update: 226.377ms
Camera Update: 218.293ms
Camera Update: 217.807ms
Camera Update: 217.28ms
Camera Update: 217.498ms
Camera Update: 217.151ms
Camera Update: 215.762ms
Camera Update: 216.684ms
Camera Update: 218.78ms
Camera Update: 219.693ms
Camera Update: 218.943ms
Camera Update: 217.011ms
Camera Update: 216.935ms
Camera Update: 218.397ms
Camera Update: 216.872ms
Camera Update: 216.411ms
Camera Update: 23417.1ms // Click and drag
Camera Update: 246.824ms
Camera Update: 237.128ms
```

Levi-Armstrong, 2023-01-24:
Loading this environment below only takes 11ms and click drag takes 4ms.

[image]

```
Camera Update: 0.264061ms
Camera Update: 0.312692ms
Camera Update: 11.3222ms // Load model
Camera Update: 1.83305ms
Camera Update: 1.43624ms
Camera Update: 3.26565ms
Camera Update: 3.44781ms
Camera Update: 3.26614ms
Camera Update: 3.48336ms
Camera Update: 3.41699ms
Camera Update: 3.5079ms
Camera Update: 3.38546ms
Camera Update: 3.45223ms
Camera Update: 3.23643ms
Camera Update: 4.00466ms // Click and drag
Camera Update: 2.97609ms
Camera Update: 3.42399ms
Camera Update: 3.08402ms
Camera Update: 3.42905ms
Camera Update: 3.25529ms
Camera Update: 3.16946ms
```

iche033, 2023-01-24:
hmm I created a minimal example to test this but still haven't been able to reproduce the long pause and camera updates. 

Can you give the following test world a try and see if you get the same issue?

```sh
ign gazebo -v 4 test_dae.sdf
```

<details><summary>test_dae.sdf</summary>

```
<?xml version="1.0" ?>

<sdf version="1.6">
  <world name="test_dae">

    <gui fullscreen="0">

      <!-- 3D scene -->
      <plugin filename="MinimalScene" name="3D View">
        <ignition-gui>
          <title>3D View</title>
          <property type="bool" key="showTitleBar">false</property>
          <property type="string" key="state">docked</property>
        </ignition-gui>

        <engine>ogre2</engine>
        <scene>scene</scene>
        <ambient_light>0.4 0.4 0.4</ambient_light>
        <background_color>0.8 0.8 0.8</background_color>
        <camera_pose>-6 0 6 0 0.5 0</camera_pose>
        <camera_clip>
          <near>0.25</near>
          <far>10000</far>
        </camera_clip>
      </plugin>

      <!-- Plugins that add functionality to the scene -->
      <plugin filename="EntityContextMenuPlugin" name="Entity context menu">
        <ignition-gui>
          <property key="state" type="string">floating</property>
          <property key="width" type="double">5</property>
          <property key="height" type="double">5</property>
          <property key="showTitleBar" type="bool">false</property>
        </ignition-gui>
      </plugin>
      <plugin filename="GzSceneManager" name="Scene Manager">
        <ignition-gui>
          <property key="resizable" type="bool">false</property>
          <property key="width" type="double">5</property>
          <property key="height" type="double">5</property>
          <property key="state" type="string">floating</property>
          <property key="showTitleBar" type="bool">false</property>
        </ignition-gui>
      </plugin>
      <plugin filename="InteractiveViewControl" name="Interactive view control">
        <ignition-gui>
          <property key="resizable" type="bool">false</property>
          <property key="width" type="double">5</property>
          <property key="height" type="double">5</property>
          <property key="state" type="string">floating</property>
          <property key="showTitleBar" type="bool">false</property>
        </ignition-gui>
      </plugin>
      <plugin filename="CameraTracking" name="Camera Tracking">
        <ignition-gui>
          <property key="resizable" type="bool">false</property>
          <property key="width" type="double">5</property>
          <property key="height" type="double">5</property>
          <property key="state" type="string">floating</property>
[... 240 lines cut ...]
          </geometry>
          <material>
            <ambient>0.8 0.8 0.8 1</ambient>
            <diffuse>0.8 0.8 0.8 1</diffuse>
            <specular>0.8 0.8 0.8 1</specular>
          </material>
        </visual>
      </link>
    </model>

    <include>
      <static>true</static>
      <name>test_Dae</name>
      <pose>0 0 0 0 0 0</pose>
      <uri>https://fuel.gazebosim.org/1.0/iche033/models/test_dae</uri>
    </include>

  </world>
</sdf>

```

</details>

Launching the world will download this [test_dae model](https://app.gazebosim.org/iche033/fuel/models/test_dae) which consists of the 3 daes that I got from your repo (compliance device and spindle; spindle's `base_link.dae` is renamed to `spindle_base_link`). Maybe I'm doing something different.

Levi-Armstrong, 2023-01-24:
I ran the command but it prints out the help info. Do I need to install something for this command to work?
```
The 'ign' command provides a command line interface to the ignition tools.

  ign <command> [options]

List of available commands:

  help:          Print this help text.
  gui:           Launch graphical interfaces.
  log:           Record or playback topics.
  topic:         Print information about topics.
  service:       Print information about services.
  sdf:           Utilities for SDF files.
  msg:           Print information about messages.
  fuel:          Manage simulation resources.
  plugin:        Print information about plugins.

Options:

  --force-version <VERSION>  Use a specific library version.
  --versions                 Show the available versions.
  --commands                 Show the available commands.
Use 'ign help <command>' to print help for a command.
```

iche033, 2023-01-24:
oh I just assumed you're running Fortress and have gazebo already installed. You'll need `libignition-gazebo6-dev`. 
So you are using gz-rendering in your own custom application?

Levi-Armstrong, 2023-01-24:
I will give it a try. It looks like these models had a lot of objects. One had +80 and the 150. I went through and cleaned up each of the models and made them water tight where each one has around 8 objects in the dae. After cleaning things up everything loads fast and the update rate is around 10ms and the load takes around 85ms.

Levi-Armstrong, 2023-01-25:
After install I get the following error. I did a quick google search but nothing worked. I mostly use ign/gz libraries directly in my own application so not familiar with a lot of the tools. 

```
[Wrn] [ign.cc:99] Fuel world download failed because Fetch failed. Other errors
Unable to find or download file 
```

Levi-Armstrong, 2023-01-25:
So I tried downloading the model locally and navigating to the folder and running the following command, but the GUI pops up and then promptly closes.

```
~/Documents/test_dae$ ign gazebo -v 4 model.sdf 
[Msg] Ignition Gazebo GUI    v6.12.0
[Dbg] [ign.cc:161] Subscribing to [/gazebo/starting_world].
[Dbg] [ign.cc:163] Waiting for a world to be set from the GUI...
[Dbg] [Gui.cc:253] Waiting for subscribers to [/gazebo/starting_world]...
[Msg] Received world [model.sdf] from the GUI.
[Dbg] [ign.cc:167] Unsubscribing from [/gazebo/starting_world].
[Msg] Ignition Gazebo Server v6.12.0
[Msg] Loading SDF world file[/home/larmstrong/Documents/test_dae/model.sdf].
[Msg] Serving world names on [/gazebo/worlds]
[Msg] Resource path add service on [/gazebo/resource_paths/add].
[Msg] Resource path get service on [/gazebo/resource_paths/get].
[Msg] Resource path resolve service on [/gazebo/resource_paths/resolve].
[Msg] Resource paths published on [/gazebo/resource_paths].
[Msg] Server control service on [/server_control].
[Dbg] [ign.cc:410] Shutting down ign-gazebo-server
[Dbg] [SignalHandler.cc:141] Received signal[2].
[Dbg] [Application.cc:92] Initializing application.
[GUI] [Dbg] [Application.cc:555] Create main window
[GUI] [Dbg] [PathManager.cc:66] Requesting resource paths through [/gazebo/resource_paths/get]
[GUI] [Wrn] [Application.cc:797] [QT] file::/Gazebo/GazeboDrawer.qml:147:3: QML Dialog: Binding loop detected for property "implicitHeight"
[GUI] [Wrn] [Application.cc:797] [QT] file::/Gazebo/GazeboDrawer.qml:147:3: QML Dialog: Binding loop detected for property "implicitHeight"
[GUI] [Dbg] [Application.cc:140] Terminating application.
```

Levi-Armstrong, 2023-01-25:
Okay, after creating the file below and using this file to load everything worked. Using this I did not see a pause. 

In my case I am not using gazebo or qml, but instead I have ported the qml render widget to Qt Widget [here](https://github.com/tesseract-robotics/tesseract_qt/blob/main/rendering/src/render_widget.cpp) specifically the QOpenGLWidget. It currently is not running the update in another thread like how the qml component is doing so the update call is happening in the main gui thread which may be causing the issue. If you have time to take a look a look at the render widget let me know if you see any issues. My hope is once I have worked out the issues is create an example to add to gui repository.

``` 
<?xml version="1.0" ?>
<sdf version="1.6">
  <world name="test_world">
    <include>
      <static>true</static>
      <name>test</name>
      <pose>0 0 0 0 0 0</pose>
      <uri>https://fuel.gazebosim.org/1.0/iche033/models/test_dae</uri>
    </include>
  </world>
</sdf>
```

iche033, 2023-01-25:
I see. Having a separate rendering thread from gui is a good idea. Another thing to try is throttle camera updates to something like 60Hz. 

Another minor optimization is that the `screenToScene` function (specifically the `ClosestPoint` function) can be expensive. I think we are unnecessary calling this function too often in gz-gui but we can't remove it as some gui plugins depend on it. If you don't need to broadcast  hover events, you can try avoiding that call.

Levi-Armstrong, 2023-01-29:
I have made the update rate configurable with the default set to 60Hz. I am going to create a minimal example only using the API and no GUI to see if the update take a long time.

azeey, 2024-05-31:
From reading through the thread, this doesn't seem to be an issue with gz-rendering. I'll go ahead and close the issue, but feel free to reopen.

closed 2024-05-31 by azeey

Levi-Armstrong, 2024-06-02:
I think there is still an issue with gz-rendering because the same mesh was not an issue when using rviz, but not sure what the difference is.

azeey, 2024-06-03:
I was under the impression that it is working fine in Gazebo (`ign gazebo` or `gz sim`), but not in your application that uses QtWidget. Is it possible to reproduce the issue with just `gz-rendering` (without Qt) or in Gazebo? Otherwise, I'm not sure how we debug the problem.