gazebosim/gz-rendering #846: Transparency with glsl shader

jrutgeer, 2023-05-09:
Hi,

I asked the question first at [answers.gazebosim.org](https://answers.gazebosim.org/question/29041/transparency-with-custom-shader/) but as it gathers little views and no answer, I suppose it is too much of a niche question for that forum:

I'm trying to control transparency (of parts of a mesh) with a fragment shader. 

As a first test, I tried to adapt the [deformable sphere example](https://app.gazebosim.org/iche033/fuel/models/deformable_sphere_no_shadows) by changing the alpha value in `deform_fs.glsl`:

i.e. change

```
fragColor = vec4(finalColor, 1);
```

into

```
fragColor = vec4(finalColor, 0.5);
```

But it remains opaque.

Defining another visual in the sdf without shaders but with element `<transparency>0.5</transparency>` does yield a transparent object. Adding the `<transparency>` element to the visual with the shaders has no effect.

Is there a way to do this in the fragment shader?

iche033, 2023-05-09:
we ran into this problem before as well. It's an issue in gz-rendering. If you're building from source, this quick patch enables  alpha blending (`gz-rendering7` branch):

```diff
diff --git a/ogre2/src/Ogre2Material.cc b/ogre2/src/Ogre2Material.cc
index 240e9164..46415802 100644
--- a/ogre2/src/Ogre2Material.cc
+++ b/ogre2/src/Ogre2Material.cc
@@ -1535,6 +1535,9 @@ void Ogre2Material::SetFragmentShader(const std::string &_path)
 
   auto mat = this->Material();
   auto pass = mat->getTechnique(0u)->getPass(0);
+  Ogre::HlmsBlendblock block;
+  block.setBlendType(Ogre::SBT_TRANSPARENT_ALPHA);
+  pass->setBlendblock(block);
   pass->setFragmentProgram(fragmentShader->getName());
   mat->compile();
   mat->load();
```

I need to make sure this doesn't break other things first before apply this fix.

jrutgeer, 2023-05-16:
@iche033 Thanks for the patch!

I got round to test it, it doesn't fully work as expected though:

Cfr. image below:
- The box is transparent due to its material in the SDF:
  >  `<material>`
  >  `       <ambient>0.3  0.3  0.3  0.8</ambient>`
  >  `       <diffuse>0.7  0.7  0.7  0.8</diffuse>`
  >  `       <specular>1 1 1 0.8</specular>`
  >  `</material>`

- The sphere has no `<material>` tag in the SDF, but has the shader param plugin with `fragColor = vec4(finalColor, 0.4f);` in the fragment shader.

Sphere in front of box:
It's not clear to me why the box is not visible through the sphere, whereas the ground plane is.

[image]

Box in front of sphere:

[image]

On the other hand, if neither box nor sphere have a `<material>` tag (i.e. both have the shader param plugin) then it seems to work correctly:

[image]

Further remarks:
- Based on [this](https://ogrecave.github.io/ogre-next/api/2.2/struct_ogre_1_1_hlms_blendblock.html#ad34f6d46eece2606c09a6eea73de4591) I tried to add `block.setForceTransparentRenderOrder(true);` before `pass->setBlendblock(block);` but this does not seem to have an effect,
- [This post](https://forums.ogre3d.org/viewtopic.php?p=520887#p520887) seems to imply that you can't just create a `HlmsBlendblock` but that you have to reauest it via `via HlmsManager::getBlendblock()`?

iche033, 2023-05-19:
I think it could be that these are 2 different types of OGRE materials and they don't mix well. The shader materials are done through the old OGRE low level material system while the materials specified through `<material>` is implemented using OGRE's [HLMS material system](https://wiki.ogre3d.org/HLMS+Materials). 

> [This post](https://forums.ogre3d.org/viewtopic.php?p=520887#p520887) seems to imply that you can't just create a HlmsBlendblock but that you have to reauest it via via HlmsManager::getBlendblock()?

gave this a quick try but unfortunately it didn't make a difference.

Hmm I'll need to look into this a bit more.