gazebosim/gz-rendering #880: Gizmo uses a transform identifier not included in `TransformAxis`

azeey, 2023-08-15:
## Environment
* OS Version: all
* Source or binary build? all

## Description
* Expected behavior:  `GizmoVisual` uses a transform identifier defined in `TransformAxis`.
* Actual behavior: As seen in https://github.com/gazebosim/gz-rendering/blob/e58870ce3ea82711fc42777709e2170341b1eb3c/include/gz/rendering/base/BaseGizmoVisual.hh#L231, `GizmoVisual` uses `TransformAxis::TA_TRANSLATION_Z << 1` which is not documented.

iche033, 2023-08-15:
hmm the logic there is to find a unique key that's not already used by `this->visuals`. I'll add some comments to it

iche033, 2024-01-23:
Addressed by #881, and #882 (see related comment: https://github.com/gazebosim/gz-sim/pull/2051#discussion_r1296506004)

closed 2024-01-23 by iche033