gazebosim/gz-rendering #884: OGRE2 not found

championAI, 2023-08-23:
when cmake , get output:
-- Looking for OGRE using the name: OGRE2
--   ! OGRE2 not found
-- Looking for OGRE using the name: OGRE-Next
--   ! OGRE-Next not found
-- Looking for GzOGRE2 - not found

but i did "sudo apt install libogre-2.2-dev" already.

ahcorde, 2023-08-23:
Hi @championAI,

I recommend asking this question on https://robotics.stackexchange.com/. Can you specify there which OS are you using and Gazebo version ? which instructions are you following ? or other relevant data there.

closed 2023-08-23 by ahcorde

championAI, 2023-08-28:
ubuntu20.04  gz-rendering7

championAI, 2023-08-28:
when "cmake .." ,
this problem arises

championAI, 2023-08-28:
-- Found OpenGL: /usr/lib/x86_64-linux-gnu/libOpenGL.so  found components: OpenGL EGL 
-- Looking for OpenGL - found

-- Checking for module 'OGRE-RTShaderSystem >= 1.9'
--   Found OGRE-RTShaderSystem , version 1.9.0
-- Checking for module 'OGRE-Terrain >= 1.9'
--   Found OGRE-Terrain , version 1.9.0
-- Checking for module 'OGRE-Overlay >= 1.9'
--   Found OGRE-Overlay , version 1.9.0
-- Checking for module 'OGRE-Paging >= 1.9'
--   Found OGRE-Paging , version 1.9.0
-- Looking for GzOGRE - found

-- Found Boost: /usr/lib/x86_64-linux-gnu/cmake/Boost-1.71.0/BoostConfig.cmake (found version "1.71.0")  
-- Looking for OGRE using the name: OGRE2
--   ! OGRE2 not found
-- Looking for OGRE using the name: OGRE-Next
--   ! OGRE-Next not found
-- Looking for GzOGRE2 - not found

amock, 2025-01-29:
I ran into the same issue while trying to compile gz-rendering from source. Installing ogre2 on my Ubuntu 24 with the following command:

```bash
sudo apt install libogre-next-dev
```

resolved it for me.