gazebosim/gz-rendering #885: Models with PBR texture maps and lightmap are dimly lit and have pale colors.

jrutgeer, 2023-08-25:
## Environment

Gazebo Garden
Ubuntu 22.04
Ogre2
Built from source

## Description

I'm running a simulation in an environment with PBR textures and light map (baked in Blender using SimpleBake). However, I can't get it to look vibrant. All colours are more greyish wrt how they render in Blender.

Not sure if this is a bug, a 'known limitation' or rather just ignorance on my side.

This is Blender (cycles):

[image]

This is gazebo sim:

[image]
(the distortion of the checker pattern is not relevant; it is due to a wrong UV mapping)

The Blender model is lit from the pendant lights (emissive white surface). The Gazebo world has the `.dae` with PBR textures, light map and at each pendant one spot light (without shadows). Both the Blender and Gazebo lights are pure white (RGB 255).

I can't share that model but I made [this test repository](https://github.com/jrutgeer/gazebo_assorted_tests/tree/main/PBR_test) that shows similarly dull colors. It holds the blender project file, as well as the derived PBR and light map images, `.dae` and `.sdf` files:

[image]

I think the [lightmap demo world](https://github.com/gazebosim/gz-sim/blob/gz-sim7/examples/worlds/lightmap.sdf) also looks rather dimly lit, so it seems that this is a more general issue?

I could obviously manually  tweak the saturation of the colours in the albedo map, but e.g. the white part of the walls is already pure white (close to max RGB values) so I can't get that to look less grey.

I also tried to add a bright light, which obviously makes the scene more clearly lit, but:
- It does not augment the saturation of the yellow colour,
- It yields uneven lighting (e.g. objects close to the light are too bright).

I also tried to change the `<ambient>` and `<ambient_light>` values in the `sdf` and `gui.config` respectively, but that does not seem to have an influence.

So I wonder if there is some way to augment the general lighting in a scene?

iche033, 2023-08-25:
The [`<ambient>`](https://github.com/gazebosim/gz-sim/blob/gz-sim7/examples/worlds/shapes.sdf#L11-L14) tag should increase overall scene brightness. I do notice differences with other game / rendering engines. Another thing to try is lower the metalness values to make objects brighter.

@ColeOSRF do you have any tips on tuning the PBR materials in gazebo to look more like other engines?

ColeOSRF, 2023-08-25:
Here is what I was able to get with some adjustments
[image]

The main issue is that your textures are 16bit. This is what causes the saturation issues with the albedo maps. They should be 8bit. For the dimness I increased the range of the point light. It's range was short to the point that it wasn't really affecting anything. You will almost certainly want some kind of dynamic light in the scene for specularity, normal maps, etc since a lightmap cannot reproduce these. A soft, large range point light without shadows is great for this. You may need more than one to create more even lighting. 

You can also play around with the light parameters like linear, constant, and quadratic to achieve a softer more even lighting. The lightmaps are just images, not hdr maps, so think of them more as shadow maps. Great for shadows and reproducing ambient occlusion, not so much for adding brightness. The dynamic lights will still be needed for to achieve a brighter environment.

Also just as an extra note. You have a lot of textures that are one solid value. Feel free to use numeric values in the SDF instead of textures in these cases. For instance <roughness>0.5</roughness> will equate to a texture map with 50% grey. 

I also lowered the ambient to zero as this can lead to the image appearing washed out and isn't usually necessary with lightmaps.

Hope this helps!

jrutgeer, 2023-08-26:
@ColeOSRF Cole Biesemeyer YOU ARE MY HERO!! :-D

Thank you so much! I would have never found that it is due to the images being 16 bit!
I had enabled 16 bit output since the SimpleBake doc states: "16 bit export should probably be enabled unless file size is an issue¨. 

Thanks also for the suggestions wrt the lighting.

And you are right about the solid value textures; the original idea was to apply real textures to all parts, but as I am alredy rather pleased with the current result I might indeed change the texture maps for numeric values.

closed 2023-08-26 by jrutgeer