gazebosim/gz-rendering #968: Issues with depth cameras

Sp0och, 2024-02-14:
## Environment
* OS Version: Ubuntu 20.04
* Source or binary build?
  Gazebo Garden source build 
gz-rendering is on branch gz-rendering7 with the latest commit being:
commit 451e34e46622ab7a8f6aa4d521ba0a1565b8fdb7 
Update examples to use metal as default graphics API on mac (#937)
    
    * Use metal as default on apple, fix compile warning, remove optix
    * Add defaultGraphicsAPI utility function
    
    ---------
    
    Signed-off-by: Ian Chen <ichen@openrobotics.org>

* If this is a GUI or sensor rendering bug, describe your GPU and rendering system. Otherwise delete this section. 
The GPU information I get when entering the sudo lshw -C display prompt is this:
description: VGA compatible controller
       product: GP106GL [Quadro P2200]
       vendor: NVIDIA Corporation
       physical id: 0
       bus info: pci@0000:02:00.0
       version: a1
       width: 64 bits
       clock: 33MHz
       capabilities: pm msi pciexpress vga_controller bus_master cap_list rom
       configuration: driver=nvidia latency=0
       resources: irq:86 memory:91000000-91ffffff memory:70000000-7fffffff memory:80000000-81ffffff ioport:2000(size=128) memory:92000000-9207ffff

    - Rendering plugin: [ogre2]. 
      - [x] Sensor rendering error. 
      - [ ] GUI rendering error. 
    - EGL headless mode: Not sure what
      - [ ] Running in EGL headless mode 
    - Generally, mention all circumstances that might affect rendering capabilities: 
      - [ ] running a [gz-ros2](https://github.com/gazebosim/ros_gz.git -b humble) and [ros2-ros](https://github.com/ros2/ros1_bridge) bridge to transfer the messages to ros
    - Rendering system info:
      - On Linux, provide the outputs of the following commands:
          ```bash
          LANG=C lspci -nn | grep VGA  # might require installing pciutils

0000:02:00.0 VGA compatible controller [0300]: NVIDIA Corporation GP106GL [Quadro P2200] [10de:1c31] (rev a1)

          echo "$DISPLAY"

:1

          LANG=C glxinfo -B | grep -i '\(direct rendering\|opengl\|profile\)'  # might require installing mesa-utils package

direct rendering: Yes
OpenGL vendor string: NVIDIA Corporation
OpenGL renderer string: Quadro P2200/PCIe/SSE2
OpenGL core profile version string: 4.6.0 NVIDIA 545.23.08
OpenGL core profile shading language version string: 4.60 NVIDIA
OpenGL core profile context flags: (none)
OpenGL core profile profile mask: core profile
OpenGL version string: 4.6.0 NVIDIA 545.23.08
OpenGL shading language version string: 4.60 NVIDIA
OpenGL context flags: (none)
OpenGL profile mask: (none)
OpenGL ES profile version string: OpenGL ES 3.2 NVIDIA 545.23.08
OpenGL ES profile shading language version string: OpenGL ES GLSL ES 3.20

          ps aux | grep Xorg

gdm         1119  0.0  0.2 25384456 67268 tty1   Sl+  Feb13   0:00 /usr/lib/xorg/Xorg vt1 -displayfd 3 -auth /run/user/125/gdm/Xauthority -background none -noreset -keeptty -verbose 3
username      1879  1.6  0.4 25518904 139304 tty2  Sl+  Feb13  15:37 /usr/lib/xorg/Xorg vt2 -displayfd 3 -auth /run/user/1000/gdm/Xauthority -background none -noreset -keeptty -verbose 3
username     52408  0.0  0.0   9348  2644 pts/19   S+   09:36   0:00 grep --color=auto Xorg

          sudo env LANG=C X -version  # if you don't have root access, try to tell the version of Xorg e.g. via package manager
[... 154 lines cut ...]
I get a different coverage of the underground. 
**Note ** as the intrinsic parameters of the depth map did not have an influence on the view and only the horizontal_fov tag had an effect I calculated this:

      FoV = 2 × arctan（w / 2f）

Which for a focal length fx = 256.666691776 pixels and an image resolution of 640x480 pixels (aka w = 640 pixels) yields a fov of 102.535°  or 1.789572604 radians. 

I am assuming that the depth cameras and normal cameras have the same model. They have the exact same pose. 

* Actual behavior: 
The Images from the depth camera and the normal camera are covering different areas. I tried to analyze it using visual editors and I am quite sure it is not just a translational shift. There is a different FOV.

## Steps to reproduce

1. Put the camera sensor in a model.sdf file and load them into any gazebo map with texture to verify.
2. Publish the sensor messages and look at them in rqt or any other tool

## Output

[image]
[image]

I apologize for the low quality map. It is hard to see but the difference can be seen from the position of the rocks. 

In case I missed anything check out my [stackoverflow post](https://robotics.stackexchange.com/questions/105347/gazebo-camera-and-depth-camera-with-same-parameters-show-different-fovs) about this

iche033, 2024-02-20:
hmm looks like it could be a bug in depth camera FOV and/or the camera intrinsics published are incorrect. 

https://github.com/gazebosim/gz-sensors/issues/363 also looks related

azeey, 2024-04-04:
There is code in `CameraSensor.cc` that handles custom intrinsic parameters: https://github.com/gazebosim/gz-sensors/blob/27239b4ac9877d87273a18010a307172899dd7a6/src/CameraSensor.cc#L319, but I don't see this in `DepthCameraSensor.cc`. The same is true for `RgbdCameraSensor`. That might explain things.

azeey, 2024-06-27:
@iche033 Can this be closed now?

iche033, 2024-06-27:
https://github.com/gazebosim/gz-sensors/pull/432 should fix the issue. Please re-open if that's not the case.

closed 2024-06-27 by iche033