gazebosim/gz-rendering #969: Fortress: Light gizmo (visual) is rendered in lidar

peci1, 2024-02-27:
## Environment
* OS Version: Ubuntu 20.04
* Source or binary build? Binary 6.16
* If this is a GUI or sensor rendering bug, describe your GPU and rendering system. Otherwise delete this section. 
    - Rendering plugin: ogre, ogre2 (both affected) 
      - [x] Sensor rendering error. 
      - [ ] GUI rendering error. 
    - EGL headless mode:
      - [ ] Running in EGL headless mode 
    - Generally, mention all circumstances that might affect rendering capabilities: 
      - [ ] running on a dual GPU machine (integrated GPU + discrete GPU)
      - [ ] running on a multi-GPU machine (it has multiple discrete GPUs)
      - [x] running on real hardware
      - [ ] running in virtual machine
      - [ ] running in Docker/Singularity
      - [ ] running remotely (e.g. via SSH)
      - [ ] running in a cloud
      - [ ] using VirtualGL, XVFB, Xdummy, XVNC or other indirect rendering utilities
      - [x] GPU is concurrently used for other tasks
        - [x] desktop acceleration
        - [ ] video decoding (i.e. a playing Youtube video)
        - [ ] video encoding
        - [ ] CUDA/ROCm computations (Tensorflow, Torch, Caffe running)
        - [ ] multiple simulators running at the same time
      - [ ] other...
    - Rendering system info:
      - On Linux, provide the outputs of the following commands:

```bash
$ LANG=C lspci -nn | grep VGA  # might require installing pciutils
06:00.0 VGA compatible controller [0300]: Advanced Micro Devices, Inc. [AMD/ATI] Renoir [1002:1636] (rev d1)
$           echo "$DISPLAY"
:0
$           LANG=C glxinfo -B | grep -i '\(direct rendering\|opengl\|profile\)'  # might require installing mesa-utils package
direct rendering: Yes
    Preferred profile: core (0x1)
    Max core profile version: 4.6
    Max compat profile version: 4.6
    Max GLES1 profile version: 1.1
    Max GLES[23] profile version: 3.2
OpenGL vendor string: AMD
OpenGL renderer string: AMD Radeon Graphics (radeonsi, renoir, LLVM 15.0.7, DRM 3.57, 6.7.5-060705-generic)
OpenGL core profile version string: 4.6 (Core Profile) Mesa 24.0.1 - kisak-mesa PPA
OpenGL core profile shading language version string: 4.60
OpenGL core profile context flags: (none)
OpenGL core profile profile mask: core profile
OpenGL version string: 4.6 (Compatibility Profile) Mesa 24.0.1 - kisak-mesa PPA
OpenGL shading language version string: 4.60
OpenGL context flags: (none)
OpenGL profile mask: compatibility profile
OpenGL ES profile version string: OpenGL ES 3.2 Mesa 24.0.1 - kisak-mesa PPA
OpenGL ES profile shading language version string: OpenGL ES GLSL ES 3.20
$           ps aux | grep Xorg
peci1       4438  1.0  0.3 1602564 99212 tty2    Sl+  úno22  73:58 /usr/lib/xorg/Xorg vt2 -displayfd 3 -auth /run/user/1000/gdm/Xauthority -background none -noreset -keeptty -verbose 3
peci1     670952  0.0  0.0  11600  2544 pts/5    S+   17:08   0:00 grep --color=auto Xorg
$           sudo env LANG=C X -version  # if you don't have root access, try to tell the version of Xorg e.g. via package manager
X.Org X Server 1.20.13
X Protocol Version 11, Revision 0
Build Operating System: linux Ubuntu
Current Operating System: Linux cras-17 6.7.5-060705-generic #202402161836 SMP PREEMPT_DYNAMIC Fri Feb 16 19:10:40 UTC 2024 x86_64
Kernel command line: BOOT_IMAGE=/boot/vmlinuz-6.7.5-060705-generic root=UUID=c8df4187-97f0-4be7-b095-39958ee965df ro quiet splash vt.handoff=7
Build Date: 29 January 2024  12:44:21PM
xorg-server 2:1.20.13-1ubuntu1~20.04.15 (For technical support please see http://www.ubuntu.com/support) 
Current version of pixman: 0.38.4
	Before reporting problems, check http://wiki.x.org
	to make sure that you have the latest version.
```
 - [ ] Please, attach the ogre.log or ogre2.log file from  `~/.gz/rendering` 

## Description
[... 141 lines cut ...]
      </link>
      <joint name='laser_j' type='fixed'>
        <child>laser</child>
        <parent>top_box</parent>
        <axis>
          <xyz>0 0 1</xyz>
        </axis>
      </joint>
      <pose>-0 -0 0.1145 0 0 -0</pose>
      <static>false</static>
      <self_collide>false</self_collide>
    </model>
  </world>
</sdf>
```

</details>

To reproduce, run `rosrun ros_ign_bridge parameter_bridge '/world/example/model/X1/link/laser/sensor/laser/scan/points@sensor_msgs/PointCloud2[ignition.msgs.PointCloudPacked'` and open up rviz with `rviz -f X1/laser/laser` and add the PointCloud2 topic.

## Output

[image]

[image]

iche033, 2024-02-27:
looks like the light visuals are created in both server and client. We'll probably need to move the code for creating light visuals to a GUI plugin so that light visuals are created in the GUI only. One option is to move code from rendering/RenderUtil.cc to `gui/plugins/visualization_capabilities/VisualizationCapabilities.cc`

peci1, 2024-02-27:
Wouldn't it be enough to set proper visibility mask?

iche033, 2024-02-27:
yeah that's my first thought as well but looking at how other visualizations are implemented right now, it may be more consistent to move the light visualization to GU. It's is a little more work though

commit 213765b3 (2024-06-03): Fix crash when visualizing Lidars with points (#1008)

Fixes: #969