gazebosim/gz-rendering #998: compile errors

heeplr, 2024-05-08:
## Environment
* OS Version: gentoo
* Source or binary build?
  source: 8.1.1 tag

<details>

```
[build.log](https://github.com/gazebosim/gz-rendering/files/15250627/build.log)
```

</details>

## Description
* Expected behavior: successful compile
* Actual behavior: compile error
 * ```error: conflicting return type specified for ‘virtual float gz::rendering::v8::OgreMovableText::getBoundingRadius()```
 * ```error: no matching function for call to ‘max(float&, Ogre::Real)’```

## Steps to reproduce
1. compile from source with gcc 13.2.1

## Output
s. Details & next comment

traversaro, 2024-05-08:
Can you provide the list of dependencies (I am not familiar with gentoo, but I guess there is some way of listing the dependencies). Thanks!

traversaro, 2024-05-08:
Also a more complete log error (for example indicating also where the error appeared) could be useful to provide hints on what is going wrong.

heeplr, 2024-05-08:
Please see the attached/embedded build.log in the "Details" collapsible.

Dependencies should also be listed in the cmake output. But iirc nothing non-gz specific besides ogre. (I'm currently afk but can provide more details later)

traversaro, 2024-05-08:
Ah cool thanks, sorry I missed that. However, I do not see any dependency version listed there. From the some flags, it seems that you are trying to use some ogre version >= 1.12 ? As far as I know, that is not tested and based on my past experience I doubt it works out of the box. For reference, I tried to get ogre 1.12 to work in conda-forge packaging (see https://github.com/conda-forge/libignition-gazebo-feedstock/issues/6#issuecomment-1580009393) and I was never be able to get it to work, so we are pinning ogre to 1.10 to get it to work. Anyhow, most of the focus now in gz-rendering is ogre-next, have you considered packaging and using that?

heeplr, 2024-05-09:
Sorry for the confusion, I was in a hurry and didn't fill the form thorougly.

I'm building againtst ogre 1.9.0, is that too low?

Here's the depgraph:

```
 * dependency graph for sci-libs/gz-rendering-8.1.1
 `--  sci-libs/gz-rendering-8.1.1  ~amd64 
   `--  media-libs/freeimage-3.18.0-r11  (media-libs/freeimage) amd64 
   `--  dev-games/ogre-1.9.0-r3  (dev-games/ogre) amd64 
   `--  dev-build/gz-cmake-3.5.3  (>=dev-build/gz-cmake-3.5.3) ~amd64 
   `--  sci-libs/gz-math-7.4.0  (>=sci-libs/gz-math-7.4.0) ~amd64 
   `--  sci-libs/gz-common-5.6.0  (>=sci-libs/gz-common-5.6.0) ~amd64 
   `--  sci-libs/gz-plugin-2.0.3  (>=sci-libs/gz-plugin-2.0.3) ~amd64 
   `--  sci-libs/gz-utils-2.2.0  (>=sci-libs/gz-utils-2.2.0) ~amd64 
   `--  app-alternatives/ninja-1  (app-alternatives/ninja) amd64 
   `--  dev-build/cmake-3.27.9  (>=dev-build/cmake-3.20.5) amd64 
```

I'll try to build ogre-next v2.3.3 and see if it gz-rendering compiles against that.

traversaro, 2024-05-09:
Sorry, I was confused by the `OGRE_VERSION_LT_1_12_0` macros and read them as `OGRE_VERSION_GT_1_12_0`. Indeed ogre 1.9.0 is the version used in Ubuntu packaging, so I am not sure what is going on. However, without ogre-next most of the gz-sim example will fail as they default to use ogre-next as rendered, so packaging also ogre-next is for sure a good choice.

traversaro, 2024-05-09:
Reading a bit better the error, it seems that the problem is that your ogre build defines `Ogre::Real` as `double`, while gz-rendering code base silently assumes that it is defined as `float` (as I guess is defined in debian, conda-forge and other packaging system builds of ogre). Probably your build of ogre 1.9 is compiled with the `OGRE_DOUBLE_PRECISION` CMake option enabled (see https://github.com/OGRECave/ogre/blob/v1-9-0/CMakeLists.txt#L393), instead of the default that is off? I guess in theory you could remove this assumptions in gz-rendering with a lot of explicit casts added, but I am not sure how easy it would be.

traversaro, 2024-05-09:
(If you could modify the issue title in something more descriptive like "Build failure with ogre 1.9.0 compiled with the `OGRE_CONFIG_DOUBLE` enabled" it would be great, thanks!

heeplr, 2024-05-09:
Ok, I disabled OGRE_CONFIG_DOUBLE (and the equivalent in bullet) and now gz-rendering 8.1.1 builds fine with ogre 1.9.0.

Thanks for your help.