gazebosim/gz-rendering #1007: INTEGRATION_load_unload_ogre_gl3plus test fails on gz_rendering-ci-pr_any-noble-amd64

iche033, 2024-05-30:
## Environment
* OS Version: Ubuntu Noble 24.04
* Source or binary build? source build, `main` branch (gz-rendering9)

    - Rendering plugin: ogre. 

## Description
`INTEGRATION_load_unload_ogre_gl3plus` test is failing on `gz_rendering-ci-pr_any-noble-amd64` CI

The test passes on Ubuntu Jammy on github actions and jenkins

## Output

CI console output

```
39:  GZ_ENGINE_TO_TEST=ogre
39:  GZ_ENGINE_BACKEND=gl3plus
39: Test timeout computed to be: 240
39: Running main() from /home/jenkins/workspace/gz_rendering-ci-pr_any-noble-amd64/gz-rendering/test/gtest_vendor/src/gtest_main.cc
39: [==========] Running 1 test from 1 test suite.
39: [----------] Global test environment set-up.
39: [----------] 1 test from LoadUnloadTest
39: [ RUN      ] LoadUnloadTest.Thread
 39/132 Test  #39: INTEGRATION_load_unload_ogre_gl3plus .................***Exception: SegFault  0.63 sec
```

Blast545, 2024-06-10:
:farmer: This is also affecting main noble jobs, see:
https://build.osrfoundation.org/job/gz_rendering-ci-main-noble-amd64/6/

iche033, 2024-07-17:
I was able to reproduce the issue on Ubuntu Noble with ogre 1.9.0 debs but I haven't found a fix for it yet. Some observations:
* Test is designed to verify rendering works in a separate thread. Crash happens when the rendering (ogre) thread exists. No errors when deleting and cleaning up gz-rendering / ogre resources.
* `gdb` did not give any useful backtraces.
* Commenting out the call to [delete ogre root](https://github.com/gazebosim/gz-rendering/blob/a7cf4f0dba349f9f6fa9a70a8f082862e387f2e7/ogre/src/OgreRenderEngine.cc#L116) made the test pass but that's not ideal.
* When testing with ogre 1.9.0 built from source (with the same cmake args as ones used to build the debs), the test passes.

iche033, 2024-08-09:
> When testing with ogre 1.9.0 built from source (with the same cmake args as ones used to build the debs), the test passes.

I missed a few compile flags last time and now have found the one that caused the crash: `-flto=auto`. Running the test with ogre 1.9 built with this flag will reliably make the test fail. It seems to be related to the `RenderSystem_GL.so` plugin. If you comment out the call to [unload](https://github.com/OGRECave/ogre/blob/v1-9-0/OgreMain/src/OgreDynLib.cpp#L118) (`dlclose`) the plugin, test does not crash when thread exists.

iche033, 2024-08-27:
merged workaround in https://github.com/gazebosim/gz-rendering/pull/1033

closed 2024-08-27 by iche033