gazebosim/gz-rendering #1018: Source Installation fails on macOS 14.5 failed on installing dependencies

qingyouzhao, 2024-07-23:
## Environment
* OS Version: macOS 14.5
* Source or binary build?
   Source build, branch: gz-rendering8

## Description
* Expected behavior: `brew install --only-dependencies gz-rendering8` should install successfully
* Actual behavior: `brew install --only-dependencies gz-rendering<#>` fails on downloading https://gitlab.gnome.org/tschoonj/gobject-introspection/-/commit/a7be304478b25271166cd92d110f251a8742d16b.diff

## Steps to reproduce

Following [installation guide](https://gazebosim.org/api/rendering/8/installation.html), for step 2 ran the command

```
brew install --only-dependencies gz-rendering8
```

## Output
The command failed with following output 
```
==> Fetching gobject-introspection
Warning: Building gobject-introspection from source as the bottle needs:
- HOMEBREW_CELLAR: /opt/homebrew/Cellar (yours is /Users/qingyou/homebrew/Cellar)
- HOMEBREW_PREFIX: /opt/homebrew (yours is /Users/qingyou/homebrew)
==> Downloading https://raw.githubusercontent.com/Homebrew/homebrew-core/75a85f205e88fe98005b9d4e1b5c2a2b0a688784/Formula/g/gobject-introspection.rb
############################################################################################################################################### 100.0%
==> Downloading https://gitlab.gnome.org/tschoonj/gobject-introspection/-/commit/a7be304478b25271166cd92d110f251a8742d16b.diff
curl: (22) The requested URL returned error: 504                                                                                    #     #    #  #
Warning: Problem : HTTP error. Will retry in 1 seconds. 3 retries left.
curl: (22) The requested URL returned error: 503                                                                                         #    #   ##
Warning: Problem : HTTP error. Will retry in 2 seconds. 2 retries left.
curl: (22) The requested URL returned error: 500                                       #           #         #         #
Warning: Problem : HTTP error. Will retry in 4 seconds. 1 retries left.
curl: (22) The requested URL returned error: 504

Error: gz-rendering8: Failed to download resource "gobject-introspection--patch"
Download failed: https://gitlab.gnome.org/tschoonj/gobject-introspection/-/commit/a7be304478b25271166cd92d110f251a8742d16b.diff
```

azeey, 2024-07-26:
hmm, we're not experiencing this on our [CI](https://build.osrfoundation.org/view/gz-harmonic/job/gz_rendering8-install_bottle-homebrew-amd64/). Could this be a temporary network issue?

qingyouzhao, 2024-07-26:
> hmm, we're not experiencing this on our [CI](https://build.osrfoundation.org/view/gz-harmonic/job/gz_rendering8-install_bottle-homebrew-amd64/). Could this be a temporary network issue?

You are right. I tried again today and the installation `brew install --only-dependencies gz-rendering8` finishes fine.

closed 2024-07-26 by qingyouzhao