gazebosim/gz-rendering #1039: How to add DynamicLines

Masa0u0, 2024-08-16:
In Gazebo Classic, you can create a DynamicLine by inheriting from gazebo::VisualPlugin and invoking gazebo::rendering::Visual::createDynamicLine. Is it possible to perform a similar operation in Gazebo Harmonic? Although the class gz::rendering::OgreDynamicLines does exist, I am unsure of the method to generate it within a scene.

iche033, 2024-08-16:
there is currently no API to create a dynamic line. The alternative is to use a [Marker](https://github.com/gazebosim/gz-rendering/blob/gz-rendering8/include/gz/rendering/Marker.hh) and set the type to `gz::rendering::MarkerType::MT_LINE_LIST` - this tells the marker to use dynamic lines. As an example, the [MarkerMananger](https://github.com/gazebosim/gz-sim/blob/gz-sim8/src/rendering/MarkerManager.cc) in gz-sim uses this to draw shapes and lines in the scene.

Masa0u0, 2024-08-16:
@iche033 
Thank you very much!

closed 2024-08-16 by iche033