gazebosim/gz-rendering #1072: UNIT_RenderingIface_TEST_ogre2_gl3plus test tries to load ogre and fails if it is not able to load it

traversaro, 2024-10-03:
## Environment
* OS Version: Ubuntu 24.04
* Source or binary build? Source https://github.com/gazebosim/gz-rendering/commit/a9af9847376267ed31592892121b03841a27a492

## Description

The UNIT_RenderingIface_TEST_ogre2_gl3plus fails if gz-rendering is compiled with ogre(1) support but no X server is available, as it actually tries to load ogre(1) plugin in https://github.com/gazebosim/gz-rendering/blob/a9af9847376267ed31592892121b03841a27a492/test/common_test/RenderingIface_TEST.cc#L126-L133 as it select the engine with index 0, that is ogre1.

### Expected behavior: 

I would expect that even if no X server is available (and so ogre plugin can't work), `DISPLAY= ctest -VV -E "_ogre_"` would be able to run.

### Actual behavior: 

With `DISPLAY= ctest -VV -E "_ogre_"` all tests pass fine, except for `UNIT_RenderingIface_TEST_ogre2_gl3plus`.

## Steps to reproduce

You can see that `UNIT_RenderingIface_TEST_ogre2_gl3plus` fails if there is no X server by running it by setting `DISPLAY` to an empty variable

1. DISPLAY= ctest -VV -R UNIT_RenderingIface_TEST_ogre2_gl3plus 
2.
3.

## Output

~~~
Running main() from /home/conda/feedstock_root/build_artifacts/gz-rendering8_1727901808682/work/test/gtest_vendor/src/gtest_main.cc
[==========] Running 3 tests from 1 test suite.
[----------] Global test environment set-up.
[----------] 3 tests from RenderingIfaceTest
[ RUN      ] RenderingIfaceTest.HasEngine
[       OK ] RenderingIfaceTest.HasEngine (0 ms)
[ RUN      ] RenderingIfaceTest.GetEngine
[Wrn] [Ogre2RenderEngine.cc:548] Unable to open display: . Trying to run in headless mode.
[Err] [RenderEngineManager.cc:484] Failed to load plugin [no_such_engine] : couldn't find shared library.
[Err] [RenderEngineManager.cc:231] Invalid render-engine index: 1000000
[Dbg] [CommonRenderingTest.hh:44] Read GZ_ENGINE_TO_TEST=ogre2
[Dbg] [CommonRenderingTest.hh:47] Read GZ_ENGINE_BACKEND=gl3plus
[Dbg] [RenderingIface.cc:135] No rendering engine is loaded yet
[Msg] Loading plugin [gz-rendering-ogre2]
[Dbg] [RenderingIface.cc:135] No rendering engine is loaded yet
[Msg] Loading plugin [no_such_engine]
[       OK ] RenderingIfaceTest.GetEngine (283 ms)
[ RUN      ] RenderingIfaceTest.RegisterEngine
[Err] [OgreRenderEngine.cc:358] Unable to open display: 
[Err] [OgreRenderEngine.cc:469] Unable to load Ogre Plugin[/home/conda/feedstock_root/build_artifacts/gz-rendering8_1727901808682/_h_env_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placeho/lib/OGRE/RenderSystem_GL]. Rendering will not be possible.Make sure you have installed OGRE properly.
[Err] [OgreRenderEngine.cc:512] unable to find OpenGL rendering system. OGRE is probably installed incorrectly. Double check the OGRE cmake output, and make sure OpenGL is enabled.

~~~

traversaro, 2024-10-03:
I give a quick shot to fix this, but using the engine name in `engine`/`unregisterEngine` does not work, and I could not find any way to quickly associate the `ogre2` string to the corresponding registered engine index.

azeey, 2025-01-07:
@iche033 / @j-rivero mind taking a look?

iche033, 2025-01-10:
I was able to reproduce the problem and created a fix in https://github.com/gazebosim/gz-rendering/pull/1098.

closed 2025-01-14 by iche033