gazebosim/gz-rendering #1107: 🧑‍🌾 Ogre gl3plus tests failing due to `unsupported engine`

Crola1702, 2025-02-13:
## Environment
* OS Version: Windows conda legacy ogre
* Source build

## Output

Reference builds:
* https://build.osrfoundation.org/job/gz_rendering-main-clowin/4/
* https://build.osrfoundation.org/job/gz_rendering-9-clowin/6
* https://build.osrfoundation.org/job/gz_rendering-8-clowin/5/

<details>
<summary>
Test regressions
</summary>

* [INTEGRATION_boundingbox_camera_ogre_gl3plus.test_ran](https://build.osrfoundation.org/job/gz_rendering-main-clowin/4/testReport/junit/(root)/INTEGRATION_boundingbox_camera_ogre_gl3plus/test_ran/)
* [INTEGRATION_depth_camera_ogre_gl3plus.test_ran](https://build.osrfoundation.org/job/gz_rendering-main-clowin/4/testReport/junit/(root)/INTEGRATION_depth_camera_ogre_gl3plus/test_ran/)
* [INTEGRATION_gpu_rays_ogre_gl3plus.test_ran](https://build.osrfoundation.org/job/gz_rendering-main-clowin/4/testReport/junit/(root)/INTEGRATION_gpu_rays_ogre_gl3plus/test_ran/)
* [INTEGRATION_lidar_visual_ogre_gl3plus.test_ran](https://build.osrfoundation.org/job/gz_rendering-main-clowin/4/testReport/junit/(root)/INTEGRATION_lidar_visual_ogre_gl3plus/test_ran/)
* [INTEGRATION_load_unload_ogre_gl3plus.test_ran](https://build.osrfoundation.org/job/gz_rendering-main-clowin/4/testReport/junit/(root)/INTEGRATION_load_unload_ogre_gl3plus/test_ran/)
* [INTEGRATION_mesh_ogre_gl3plus.test_ran](https://build.osrfoundation.org/job/gz_rendering-main-clowin/4/testReport/junit/(root)/INTEGRATION_mesh_ogre_gl3plus/test_ran/)
* [INTEGRATION_render_pass_ogre_gl3plus.test_ran](https://build.osrfoundation.org/job/gz_rendering-main-clowin/4/testReport/junit/(root)/INTEGRATION_render_pass_ogre_gl3plus/test_ran/)
* [INTEGRATION_segmentation_camera_ogre_gl3plus.test_ran](https://build.osrfoundation.org/job/gz_rendering-main-clowin/4/testReport/junit/(root)/INTEGRATION_segmentation_camera_ogre_gl3plus/test_ran/)
* [INTEGRATION_sky_ogre_gl3plus.test_ran](https://build.osrfoundation.org/job/gz_rendering-main-clowin/4/testReport/junit/(root)/INTEGRATION_sky_ogre_gl3plus/test_ran/)
* [ThermalCameraTest.ThermalCameraBoxesHeatSignature](https://build.osrfoundation.org/job/gz_rendering-main-clowin/4/testReport/junit/(root)/ThermalCameraTest/ThermalCameraBoxesHeatSignature/)
* [INTEGRATION_thermal_camera_ogre_gl3plus.test_ran](https://build.osrfoundation.org/job/gz_rendering-main-clowin/4/testReport/junit/(root)/INTEGRATION_thermal_camera_ogre_gl3plus/test_ran/)
* [INTEGRATION_waves_ogre_gl3plus.test_ran](https://build.osrfoundation.org/job/gz_rendering-main-clowin/4/testReport/junit/(root)/INTEGRATION_waves_ogre_gl3plus/test_ran/)
* [REGRESSION_reload_engine_ogre_gl3plus.test_ran](https://build.osrfoundation.org/job/gz_rendering-main-clowin/4/testReport/junit/(root)/REGRESSION_reload_engine_ogre_gl3plus/test_ran/)
* [UNIT_ArrowVisual_TEST_ogre_gl3plus.test_ran](https://build.osrfoundation.org/job/gz_rendering-main-clowin/4/testReport/junit/(root)/UNIT_ArrowVisual_TEST_ogre_gl3plus/test_ran/)
* [UNIT_AxisVisual_TEST_ogre_gl3plus.test_ran](https://build.osrfoundation.org/job/gz_rendering-main-clowin/4/testReport/junit/(root)/UNIT_AxisVisual_TEST_ogre_gl3plus/test_ran/)
* [UNIT_BoundingBoxCamera_TEST_ogre_gl3plus.test_ran](https://build.osrfoundation.org/job/gz_rendering-main-clowin/4/testReport/junit/(root)/UNIT_BoundingBoxCamera_TEST_ogre_gl3plus/test_ran/)
* [UNIT_COMVisual_TEST_ogre_gl3plus.test_ran](https://build.osrfoundation.org/job/gz_rendering-main-clowin/4/testReport/junit/(root)/UNIT_COMVisual_TEST_ogre_gl3plus/test_ran/)
* [UNIT_Camera_TEST_ogre_gl3plus.test_ran](https://build.osrfoundation.org/job/gz_rendering-main-clowin/4/testReport/junit/(root)/UNIT_Camera_TEST_ogre_gl3plus/test_ran/)
* [UNIT_Capsule_TEST_ogre_gl3plus.test_ran](https://build.osrfoundation.org/job/gz_rendering-main-clowin/4/testReport/junit/(root)/UNIT_Capsule_TEST_ogre_gl3plus/test_ran/)
* [UNIT_GaussianNoisePass_TEST_ogre_gl3plus.test_ran](https://build.osrfoundation.org/job/gz_rendering-main-clowin/4/testReport/junit/(root)/UNIT_GaussianNoisePass_TEST_ogre_gl3plus/test_ran/)
* [UNIT_GizmoVisual_TEST_ogre_gl3plus.test_ran](https://build.osrfoundation.org/job/gz_rendering-main-clowin/4/testReport/junit/(root)/UNIT_GizmoVisual_TEST_ogre_gl3plus/test_ran/)
* [UNIT_GlobalIllumination_TEST_ogre_gl3plus.test_ran](https://build.osrfoundation.org/job/gz_rendering-main-clowin/4/testReport/junit/(root)/UNIT_GlobalIllumination_TEST_ogre_gl3plus/test_ran/)
* [UNIT_Grid_TEST_ogre_gl3plus.test_ran](https://build.osrfoundation.org/job/gz_rendering-main-clowin/4/testReport/junit/(root)/UNIT_Grid_TEST_ogre_gl3plus/test_ran/)
* [UNIT_Heightmap_TEST_ogre_gl3plus.test_ran](https://build.osrfoundation.org/job/gz_rendering-main-clowin/4/testReport/junit/(root)/UNIT_Heightmap_TEST_ogre_gl3plus/test_ran/)
* [UNIT_InertiaVisual_TEST_ogre_gl3plus.test_ran](https://build.osrfoundation.org/job/gz_rendering-main-clowin/4/testReport/junit/(root)/UNIT_InertiaVisual_TEST_ogre_gl3plus/test_ran/)
* [UNIT_LensFlarePass_TEST_ogre_gl3plus.test_ran](https://build.osrfoundation.org/job/gz_rendering-main-clowin/4/testReport/junit/(root)/UNIT_LensFlarePass_TEST_ogre_gl3plus/test_ran/)
* [UNIT_LidarVisual_TEST_ogre_gl3plus.test_ran](https://build.osrfoundation.org/job/gz_rendering-main-clowin/4/testReport/junit/(root)/UNIT_LidarVisual_TEST_ogre_gl3plus/test_ran/)
* [UNIT_LightVisual_TEST_ogre_gl3plus.test_ran](https://build.osrfoundation.org/job/gz_rendering-main-clowin/4/testReport/junit/(root)/UNIT_LightVisual_TEST_ogre_gl3plus/test_ran/)
* [UNIT_Light_TEST_ogre_gl3plus.test_ran](https://build.osrfoundation.org/job/gz_rendering-main-clowin/4/testReport/junit/(root)/UNIT_Light_TEST_ogre_gl3plus/test_ran/)
* [UNIT_Marker_TEST_ogre_gl3plus.test_ran](https://build.osrfoundation.org/job/gz_rendering-main-clowin/4/testReport/junit/(root)/UNIT_Marker_TEST_ogre_gl3plus/test_ran/)
* [UNIT_Material_TEST_ogre_gl3plus.test_ran](https://build.osrfoundation.org/job/gz_rendering-main-clowin/4/testReport/junit/(root)/UNIT_Material_TEST_ogre_gl3plus/test_ran/)
* [UNIT_MeshDescriptor_TEST_ogre_gl3plus.test_ran](https://build.osrfoundation.org/job/gz_rendering-main-clowin/4/testReport/junit/(root)/UNIT_MeshDescriptor_TEST_ogre_gl3plus/test_ran/)
* [UNIT_Mesh_TEST_ogre_gl3plus.test_ran](https://build.osrfoundation.org/job/gz_rendering-main-clowin/4/testReport/junit/(root)/UNIT_Mesh_TEST_ogre_gl3plus/test_ran/)
* [UNIT_MoveToHelper_TEST_ogre_gl3plus.test_ran](https://build.osrfoundation.org/job/gz_rendering-main-clowin/4/testReport/junit/(root)/UNIT_MoveToHelper_TEST_ogre_gl3plus/test_ran/)
* [UNIT_Node_TEST_ogre_gl3plus.test_ran](https://build.osrfoundation.org/job/gz_rendering-main-clowin/4/testReport/junit/(root)/UNIT_Node_TEST_ogre_gl3plus/test_ran/)
* [UNIT_OrbitViewController_TEST_ogre_gl3plus.test_ran](https://build.osrfoundation.org/job/gz_rendering-main-clowin/4/testReport/junit/(root)/UNIT_OrbitViewController_TEST_ogre_gl3plus/test_ran/)
* [UNIT_OrthoViewController_TEST_ogre_gl3plus.test_ran](https://build.osrfoundation.org/job/gz_rendering-main-clowin/4/testReport/junit/(root)/UNIT_OrthoViewController_TEST_ogre_gl3plus/test_ran/)
* [UNIT_ParticleEmitter_TEST_ogre_gl3plus.test_ran](https://build.osrfoundation.org/job/gz_rendering-main-clowin/4/testReport/junit/(root)/UNIT_ParticleEmitter_TEST_ogre_gl3plus/test_ran/)
* [UNIT_Projector_TEST_ogre_gl3plus.test_ran](https://build.osrfoundation.org/job/gz_rendering-main-clowin/4/testReport/junit/(root)/UNIT_Projector_TEST_ogre_gl3plus/test_ran/)
* [UNIT_RayQuery_TEST_ogre_gl3plus.test_ran](https://build.osrfoundation.org/job/gz_rendering-main-clowin/4/testReport/junit/(root)/UNIT_RayQuery_TEST_ogre_gl3plus/test_ran/)
* [UNIT_RenderEngineManager_TEST_ogre_gl3plus.test_ran](https://build.osrfoundation.org/job/gz_rendering-main-clowin/4/testReport/junit/(root)/UNIT_RenderEngineManager_TEST_ogre_gl3plus/test_ran/)
* [UNIT_RenderPassSystem_TEST_ogre_gl3plus.test_ran](https://build.osrfoundation.org/job/gz_rendering-main-clowin/4/testReport/junit/(root)/UNIT_RenderPassSystem_TEST_ogre_gl3plus/test_ran/)
* [UNIT_RenderTarget_TEST_ogre_gl3plus.test_ran](https://build.osrfoundation.org/job/gz_rendering-main-clowin/4/testReport/junit/(root)/UNIT_RenderTarget_TEST_ogre_gl3plus/test_ran/)
* [UNIT_RenderingIface_TEST_ogre_gl3plus.test_ran](https://build.osrfoundation.org/job/gz_rendering-main-clowin/4/testReport/junit/(root)/UNIT_RenderingIface_TEST_ogre_gl3plus/test_ran/)
[... cut ...]
* [UNIT_Visual_TEST_ogre_gl3plus.test_ran](https://build.osrfoundation.org/job/gz_rendering-main-clowin/4/testReport/junit/(root)/UNIT_Visual_TEST_ogre_gl3plus/test_ran/)
* [UNIT_WireBox_TEST_ogre_gl3plus.test_ran](https://build.osrfoundation.org/job/gz_rendering-main-clowin/4/testReport/junit/(root)/UNIT_WireBox_TEST_ogre_gl3plus/test_ran/)

</details>

All the failed tests have the following environment variables:
```
Environment variables: 
  GZ_ENGINE_TO_TEST=ogre
  GZ_ENGINE_BACKEND=gl3plus
```

Some of them fail when loading `gz-rendering-ogre` plugins. Others show `Engine 'ogre' unsupported`

iche033, 2025-02-13:
> Some of them fail when loading gz-rendering-ogre plugins. Others show Engine 'ogre' unsupported

I briefly looked into this before. I think the tests are crashing on exit. For tests that print out the unsupported warnings, they actually do also load the `ogre` engine but do not execute the body of the test, and then proceed to crash on exit. I think there is an issue unloading the engine.

Crola1702, 2025-03-27:
I think this was solved recently, as the problem is not present anymore

iche033, 2025-03-27:
oh interesting, I haven't seen any related changes addressing this recently. Maybe we are using different win CI machine / graphics drivers. In any case, feel free to re-open if you see this happen again, thanks.

j-rivero, 2025-12-04:
> It didn't make it green, the error is still happening

wow, we have a ghost. Comparing the logs did not get us any hint about a difference?