gazebosim/gz-rendering #1109: 🧑‍🌾 Gz-rendering6-clwin ogre2 integration tests failing

Crola1702, 2025-02-18:
## Environment
* OS Version: Windows conda legacy
* Source gz-rendering6

Reference build:
* https://build.osrfoundation.org/job/gz_rendering-ign-rendering6-clwin/6/

Test regressions:
* [DepthCamera/DepthCameraTest.DepthCameraBoxes/ogre2](https://build.osrfoundation.org/job/gz_rendering-ign-rendering6-clwin/6/testReport/junit/(root)/DepthCamera_DepthCameraTest/DepthCameraBoxes_ogre2/)
* [ThermalCamera/ThermalCameraTest.ThermalCameraBoxesUniformTemp/ogre2](https://build.osrfoundation.org/job/gz_rendering-ign-rendering6-clwin/6/testReport/junit/(root)/ThermalCamera_ThermalCameraTest/ThermalCameraBoxesUniformTemp_ogre2/)
* [ThermalCamera/ThermalCameraTest.ThermalCameraBoxesHeatSignature/ogre2](https://build.osrfoundation.org/job/gz_rendering-ign-rendering6-clwin/6/testReport/junit/(root)/ThermalCamera_ThermalCameraTest/ThermalCameraBoxesHeatSignature_ogre2/)
* [ThermalCamera/ThermalCameraTest.ThermalCameraBoxesUniformTemp8Bit/ogre2](https://build.osrfoundation.org/job/gz_rendering-ign-rendering6-clwin/6/testReport/junit/(root)/ThermalCamera_ThermalCameraTest/ThermalCameraBoxesUniformTemp8Bit_ogre2/)
* [SceneFactory/SceneFactoryTest.MaterialMemoryLeak/ogre2](https://build.osrfoundation.org/job/gz_rendering-ign-rendering6-clwin/6/testReport/junit/(root)/SceneFactory_SceneFactoryTest/MaterialMemoryLeak_ogre2/)
* [SceneFactory/SceneFactoryTest.VisualMemoryLeak/ogre2](https://build.osrfoundation.org/job/gz_rendering-ign-rendering6-clwin/6/testReport/junit/(root)/SceneFactory_SceneFactoryTest/VisualMemoryLeak_ogre2/)

All of them show mismatch between values:

<details>
<summary>
depth_camera
</summary>

```
C:\J\workspace\gz_rendering-ign-rendering6-clwin\ws\gz-rendering\test\integration\depth_camera.cc:314
Expected equality of these values:
  minVal
    Which is: -inf
  scan[mid]
    Which is: inf
```

</details>

<details>
<summary>
thermal_camera
</summary>

```
C:\J\workspace\gz_rendering-ign-rendering6-clwin\ws\gz-rendering\test\integration\thermal_camera.cc:242
The difference between boxTemp and temp is 16, which exceeds boxTempRange, where
boxTemp evaluates to 310,
temp evaluates to 294, and
boxTempRange evaluates to 3.

...

C:\J\workspace\gz_rendering-ign-rendering6-clwin\ws\gz-rendering\test\integration\thermal_camera.cc:225
The difference between boxTemp and thermalData[mid] * linearResolution is 49.770004272460938, which exceeds boxTempRange, where
boxTemp evaluates to 150,
thermalData[mid] * linearResolution evaluates to 100.22999572753906, and
boxTempRange evaluates to 3.

...

C:\J\workspace\gz_rendering-ign-rendering6-clwin\ws\gz-rendering\test\integration\thermal_camera.cc:431
The difference between boxTemp and temp is 16, which exceeds boxTempRange, where
boxTemp evaluates to 310,
temp evaluates to 294, and
boxTempRange evaluates to 3.

```

</details>

<details>
<summary>
scene_factory
</summary>

```
C:\J\workspace\gz_rendering-ign-rendering6-clwin\ws\gz-rendering\test\performance\scene_factory.cc:127
Expected: (resPercentChange) < (resMaxPercentChange), actual: -nan(ind) vs 3
```

</details>

Crola1702, 2025-04-21:
Friendly ping @iche033

iche033, 2025-04-29:
https://github.com/gazebosim/gz-rendering/pull/1126 disables a couple of tests on `main` for now