gazebosim/gz-rendering #1112: Add a tutorial for the frustum_visual example

iche033, 2025-03-10:
A [FrustumVisual](https://github.com/gazebosim/gz-rendering/blob/gz-rendering9/include/gz/rendering/FrustumVisual.hh) class was added in https://github.com/gazebosim/gz-rendering/pull/1095 for visualizing the frustum of camera sensors. An example exists in [examples/frustum_visual]( https://github.com/gazebosim/gz-rendering/tree/gz-rendering9/examples/frustum_visual). We should add a tutorial explaining the example, very much like other tutorials in the [tutorials](https://github.com/gazebosim/gz-rendering/tree/gz-rendering9/tutorials) dir.

## Implementation suggestion

1. Add a frustum visual tutorial in the tutorials direction
1. In the tutorial, provide steps for building the example, and the expected output (screenshot).
1. Add explanations for the code, specifically for the [API calls](https://github.com/gazebosim/gz-rendering/blob/597d8892d7478b89f7f794c176f737578dd9c40a/examples/frustum_visual/Main.cc#L103-L106) needed to construct a FrustumVisual:
1. Once done, the tutorial needs to be added to [tutorials.md.in](https://github.com/gazebosim/gz-rendering/blob/main/tutorials.md.in)

SuperGops7, 2025-03-16:
Hi @iche033 , I'd love to work on this. Can this be assigned to me?

ahcorde, 2025-03-17:
Hi @SuperGops7

I assigned the issue. Feel free to include here the PR

lakhmanisahil, 2026-01-09:
Hey @SuperGops7, are you still working on this issue? If it’s available, I’d be happy to take it over.
@ahcorde, please feel free to assign it to me.