gazebosim/gz-rendering #1113: Spell check the documentation

ahcorde, 2025-03-18:
As part of our spring cleaning, we want to detect spell issues in our documentation and fix it when possible. Here's a potential plan to tackle this issue:

1. We suggest to start checking out the main branch.
1. Identify the files that contain documentation. These are typically `.md` (Markdown) or `.md.in` (a Markdown file to be generated by `CMake`).
1. Install in your system a tool to check for spelling errors (e.g.: `aspell` in Ubuntu).
1. Run the spell checks in all the documents. Most of the catches will be false positives due to words like `Protobuf`, `CMake` and other words that aren't typically in an English dictionary. Ignore all of these until you catch a real spell problem.
1. Fix it. Usually these programs recommend a proper word or you can later manually modify the file with the correct English term.
1. Create a pull request with your fixes. [This guide](https://gazebosim.org/docs/all/contributing/#contributing-code) can be helpful. Don't be shy, we'll be kind during the process and help you if you don't have much experience with Gazebo yet or the instructions are not clear enough.

XueSongTap, 2025-03-19:
Can I work on this ?

ahcorde, 2025-03-19:
Sure @XueSongTap the issue is assigned to you