gazebosim/gz-rendering #1117: 🧑‍🌾  clo-win `MeshTest.NormalMapWithoutTexCoord` SEH Exception

Crola1702, 2025-03-27:
## Environment
* OS Version: Windows Conda Legacy Ogre
* Source gz-rendering harmonic or newer

## Description
Reference builds:
* [gz_rendering-main-clowin#15](https://build.osrfoundation.org/job/gz_rendering-main-clowin/15)
* [gz_rendering-9-clowin#18](https://build.osrfoundation.org/job/gz_rendering-9-clowin/18)
* [gz_rendering-8-clowin#18](https://build.osrfoundation.org/job/gz_rendering-8-clowin/18)

Test regression:
* [MeshTest.NormalMapWithoutTexCoord](https://build.osrfoundation.org/job/gz_rendering-main-clowin/15/testReport/(root)/MeshTest/NormalMapWithoutTexCoord/)

Log output:
```
35: [ RUN      ] MeshTest.NormalMapWithoutTexCoord
35: (2025-03-24 13:20:33.142) [debug] [CommonRenderingTest.hh:44] Read GZ_ENGINE_TO_TEST=ogre
35: (2025-03-24 13:20:33.143) [debug] [CommonRenderingTest.hh:47] Read GZ_ENGINE_BACKEND=gl3plus
35: (2025-03-24 13:20:33.143) [info] [RenderEngineManager.cc:467] Loading plugin [gz-rendering-ogre]
35: unknown file: error: SEH exception with code 0xc0000005 thrown in the test body.
35: [  FAILED  ] MeshTest.NormalMapWithoutTexCoord (809 ms)
```

iche033, 2025-04-29:
https://github.com/gazebosim/gz-rendering/pull/1126 disables a couple of tests on main for now

iche033, 2025-08-12:
While doing a forward port from 8 -> 9 in https://github.com/gazebosim/gz-rendering/pull/1155, I found that this test started started failing with `ogre` on Noble github actions in `gz-rendering9`. Interestingly test passes on Jammy.

miguelgonrod, 2026-02-16:
Haven't happened for more than 30 days, should we close this issue?