gazebosim/gz-rendering #1122: Segmentation camera breaks the sim when pointing at a light source

mazqtpopx, 2025-04-24:
## Environment
* OS Version: Ubuntu 24.04
* Source or binary build?
 Both, gz-rendering9
* If this is a GUI or sensor rendering bug, describe your GPU and rendering system. Otherwise delete this section. 
NVIDIA A5000
Ogre2
      - [X] Sensor rendering error. 
      - [ ] GUI rendering error. 
    - EGL headless mode: 
      - Both with/without Running in EGL headless mode 
    - Generally, mention all circumstances that might affect rendering capabilities: 
      - [x] running on a dual GPU machine (integrated GPU + discrete GPU)
      - [ ] running on a multi-GPU machine (it has multiple discrete GPUs)
      - [x] running on real hardware
      - [ ] running in virtual machine
      - [ ] running in Docker/Singularity
      - [ ] running remotely (e.g. via SSH)
      - [ ] running in a cloud
      - [ ] using VirtualGL, XVFB, Xdummy, XVNC or other indirect rendering utilities
      - [ ] GPU is concurrently used for other tasks
        - [ ] desktop acceleration
        - [ ] video decoding (i.e. a playing Youtube video)
        - [ ] video encoding
        - [ ] CUDA/ROCm computations (Tensorflow, Torch, Caffe running)
        - [ ] multiple simulators running at the same time
      - [ ] other...
    - Rendering system info:
      - On Linux, provide the outputs of the following commands:
          ```bash
          LANG=C lspci -nn | grep VGA  # might require installing pciutils
          echo "$DISPLAY"
          LANG=C glxinfo -B | grep -i '\(direct rendering\|opengl\|profile\)'  # might require installing mesa-utils package
          ps aux | grep Xorg
          sudo env LANG=C X -version  # if you don't have root access, try to tell the version of Xorg e.g. via package manager
          ```
      - On Windows, run `dxdiag` and report the GPU-related information.
      - On Mac OS, open a terminal and type `system_profiler SPDisplaysDataType`. Copy the output here.
        
    - [ ] Please, attach the ogre.log or ogre2.log file from  `~/.gz/rendering` 

<details>

```
# paste log here
```
[gz-1] [Err] [Ogre2GzHlmsPbsPrivate.cc:280] A module is trying to render an object without specifying a parameter. Please report this bug at https://github.com/gazebosim/gz-rendering/issues
[gz-1] terminate called after throwing an instance of 'Ogre::ItemIdentityException'
[gz-1]   what():  OGRE EXCEPTION(5:ItemIdentityException): Parameter at the given index was not found. in Renderable::getCustomParameter at /opt/ros/jazzy/opt/gz_ogre_next_vendor/include/OGRE-Next/OgreRenderable.h (line 294)
[gz-1] Stack trace (most recent call last) in thread 497170:
[gz-1] #25   Object "[0xffffffffffffffff]", at 0xffffffffffffffff, in 
[gz-1] #24   Object "/lib/x86_64-linux-gnu/libc.so.6", at 0x782d55f29c3b, in 
[gz-1] #23   Object "/lib/x86_64-linux-gnu/libc.so.6", at 0x782d55e9caa3, in 
[gz-1] #22   Object "/lib/x86_64-linux-gnu/libstdc++.so.6", at 0x782d50aecdb3, in 
[gz-1] #21   Object "/opt/ros/jazzy/opt/gz_sim_vendor/lib/gz-sim-8/plugins/libgz-sim-sensors-system.so", at 0x782d3ea2fd4f, in gz::sim::v8::systems::SensorsPrivate::RenderThread()
[gz-1] #20   Object "/opt/ros/jazzy/opt/gz_sim_vendor/lib/gz-sim-8/plugins/libgz-sim-sensors-system.so", at 0x782d3ea2f2cb, in gz::sim::v8::systems::SensorsPrivate::RunOnce()
[gz-1] #19   Object "/opt/ros/jazzy/opt/gz_sensors_vendor/lib/libgz-sensors8.so.8", at 0x782d3d65609e, in gz::sensors::v8::Manager::RunOnce(std::chrono::duration<long, std::ratio<1l, 1000000000l> > const&, bool)
[gz-1] #18   Object "/opt/ros/jazzy/opt/gz_sensors_vendor/lib/libgz-sensors8.so.8", at 0x782d3d655fbb, in gz::sensors::v8::Sensor::Update(std::chrono::duration<long, std::ratio<1l, 1000000000l> > const&, bool)
[gz-1] #17   Object "/opt/ros/jazzy/opt/gz_sensors_vendor/lib/libgz-sensors8-segmentation_camera.so.8", at 0x782d3e034b8c, in gz::sensors::v8::SegmentationCameraSensor::Update(std::chrono::duration<long, std::ratio<1l, 1000000000l> > const&)
[gz-1] #16   Object "/opt/ros/jazzy/opt/gz_sensors_vendor/lib/libgz-sensors8-rendering.so.8", at 0x782d3d66dac5, in gz::sensors::v8::RenderingSensor::Render()
[gz-1] #15   Object "/opt/ros/jazzy/opt/gz_rendering_vendor/lib/gz-rendering-8/engine-plugins/libgz-rendering-ogre2.so", at 0x782d04422ef6, in gz::rendering::v8::Ogre2SegmentationCamera::Render()
[gz-1] #14   Object "/opt/ros/jazzy/opt/gz_ogre_next_vendor/lib/libOgreNextMain.so.2.3.3", at 0x782cffe4c4c3, in Ogre::CompositorWorkspace::_update(bool)
[gz-1] #13   Object "/opt/ros/jazzy/opt/gz_ogre_next_vendor/lib/libOgreNextMain.so.2.3.3", at 0x782cffe3cb35, in Ogre::CompositorNode::_update(Ogre::Camera const*, Ogre::SceneManager*)
[gz-1] #12   Object "/opt/ros/jazzy/opt/gz_ogre_next_vendor/lib/libOgreNextMain.so.2.3.3", at 0x782cffe57fb8, in Ogre::CompositorPassScene::execute(Ogre::Camera const*)
[gz-1] #11   Object "/opt/ros/jazzy/opt/gz_ogre_next_vendor/lib/libOgreNextMain.so.2.3.3", at 0x782cffbb52fb, in Ogre::Camera::_renderScenePhase02(Ogre::Camera const*, unsigned char, unsigned char, bool)
[gz-1] #10   Object "/opt/ros/jazzy/opt/gz_ogre_next_vendor/lib/libOgreNextMain.so.2.3.3", at 0x782cffd537c2, in Ogre::SceneManager::_renderPhase02(Ogre::Camera*, Ogre::Camera const*, unsigned char, unsigned char, bool)
[gz-1] #9    Object "/opt/ros/jazzy/opt/gz_ogre_next_vendor/lib/libOgreNextMain.so.2.3.3", at 0x782cffd0f504, in Ogre::RenderQueue::render(Ogre::RenderSystem*, unsigned char, unsigned char, bool, bool)
[gz-1] #8    Object "/opt/ros/jazzy/opt/gz_ogre_next_vendor/lib/libOgreNextMain.so.2.3.3", at 0x782cffd0e0b6, in Ogre::RenderQueue::renderGL3(Ogre::RenderSystem*, bool, bool, Ogre::HlmsCache*, Ogre::RenderQueue::RenderQueueGroup const&, Ogre::IndirectBufferPacked*, unsigned char*, unsigned char*)
[gz-1] #7    Object "/opt/ros/jazzy/opt/gz_rendering_vendor/lib/gz-rendering-8/engine-plugins/libgz-rendering-ogre2.so", at 0x782d04340fd9, in 
[gz-1] #6    Object "/lib/x86_64-linux-gnu/libstdc++.so.6", at 0x782d50aa5a6e, in 
[gz-1] #5    Object "/lib/x86_64-linux-gnu/libstdc++.so.6", at 0x782d50aa5a54, in std::terminate()
[gz-1] #4    Object "/lib/x86_64-linux-gnu/libstdc++.so.6", at 0x782d50abb0d9, in 
[gz-1] #3    Object "/lib/x86_64-linux-gnu/libstdc++.so.6", at 0x782d50aa5ff4, in 
[gz-1] #2    Object "/lib/x86_64-linux-gnu/libc.so.6", at 0x782d55e288fe, in abort
[gz-1] #1    Object "/lib/x86_64-linux-gnu/libc.so.6", at 0x782d55e4527d, in gsignal
[gz-1] #0    Object "/lib/x86_64-linux-gnu/libc.so.6", at 0x782d55e9eb2c, in pthread_kill
[gz-1] Aborted (Signal sent by tkill() 497085 1000)
[INFO] [gz-1]: process has finished cleanly [pid 497084]

## Description
Segmentation camera breaks the simulation when a light source is in frame 

* Expected behavior: the segmentation camera should not render the light source
* Actual behavior: gz crashes

## Steps to reproduce
Put a segmentation camera in environment and point at light source - gz should crash

## Output
To make sure it was the light sources crashing it I added a default colour (white) to the items crashing my simulation:
[image]
They showed up as white in the segmentation display. 

As a workaround I added the following to the fillBuffersForV2 function in Ogre2GzHlmsPbsPrivate.cc:
```
try
{
  if (!_queuedRenderable.renderable->hasCustomParameter(1u))
  {
    gzerr << "Auto‐assigning default seg‐ID to: 0" << "\n";
    _queuedRenderable.renderable->setCustomParameter(1u, Ogre::Vector4(0,0,0,1));  // or whatever default
    // return instanceIdx;
  }
  customParam = _queuedRenderable.renderable->getCustomParameter(1u);
}
catch (ItemIdentityException &)
{
...
```
This sets the mask of the objects to be black. Obviously, I set the Vector to be (1,1,1,1) to reproduce the image. I think the light materials should be set to be black in Ogre2SegmentationMaterialSwitcher.cc but that might require some more tinkering. I would be happy to have a look at a proper fix if you would like.

mazqtpopx, 2025-04-27:
NB: an easy fix for this is to set
```
      <visualize>false</visualize>
```
for any light sources in your environment when using segmentation cameras - this does not require compiling gz rendering from source

iche033, 2025-04-28:
created a fix: https://github.com/gazebosim/gz-sim/pull/2888

closed 2025-04-29 by iche033