gazebosim/gz-rendering #1134: Thermal camera does not react to updated settings once thermal texture has been created

mjacobse, 2025-05-07:
## Environment
* OS Version: Ubuntu 22.04
* Binary and source build 8.2.2

## Description
* Expected behavior: Changing the thermal camera resolution to very low values through a `.sdf` file will clearly influence the output
* Actual behavior: Thermal camera output is unaffected, no matter what resolution is set through the `.sdf` file

## Steps to reproduce

1. Open `examples/worlds/thermal_camera.sdf` in a text editor
2. Change the resolution of the 8-bit camera through the `gz-sim-thermal-sensor-system` plugin to `0.1`
3. Run `gz sim -r examples/worlds/thermal_camera.sdf`
4. Notice that the output of the 8-bit camera is unchanged

## Debugging efforts

I have noticed this when trying to use the `gz-sim-thermal-sensor-system` plugin to change the properties of thermal cameras, but I believe the root cause is in the rendering code itself. That is unless it is actually intended behaviour that the thermal camera does not react to setting changes until it is reset in some way which I was unable to find.

By tracing the calls for setting the thermal camera resolution through `gz-sim`, `gz-sensors` and `gz-rendering` I have seen that the callchain does indeed reach `BaseThermalCamera<T>::SetLinearResolution`. However, as far as I can tell `CreateThermalTexture` is only ever called once for the lifetime of the camera. That means that `this->resolution` is only read once and set for the shader: https://github.com/gazebosim/gz-rendering/blob/99031b74bd8bf63116eeaa8585025f0fe6b3edd3/ogre2/src/Ogre2ThermalCamera.cc#L887-L888
Any further calls to `BaseThermalCamera<T>::SetLinearResolution`, for example through the `gz-sim-thermal-sensor-system` plugin only set `this->resolution`, but never the constant in the shader, meaning that the thermal camera output remains unaffected. For Ogre2, the new resolution is also never reported to the `OgreThermalCameraMaterialSwitcher` that uses it for creating normalized materials. 

Note that `min_temp` and `max_temp` of the `gz-sim-thermal-sensor-system` plugin are also affected by this. It is not limited to use with this plugin though. Changing any of the settings that are only read once in `CreateThermalTexture` after the first call to `thermalCamera->Update()` in the thermal camera integration tests will not affect the result of any future calls to `thermalCamera->Update()` either.

As a simple workaround, I tried adding something like
```c++
  Ogre::Pass *pass =
    this->dataPtr->thermalMaterial->getTechnique(0)->getPass(0);
  Ogre::GpuProgramParametersSharedPtr psParams =
    pass->getFragmentProgramParameters();
  psParams->setNamedConstant("max",
      static_cast<float>(this->maxTemp));
  psParams->setNamedConstant("min",
      static_cast<float>(this->minTemp));
  psParams->setNamedConstant("resolution",
      static_cast<float>(this->resolution));

  this->dataPtr->thermalMaterialSwitcher->SetLinearResolution(this->resolution);
```
to the end of `Ogre2ThermalCamera::PreRender`, which then indeed allows me to successfuly change the settings through the `gz-sim-thermal-sensor-system` plugin. However, I am very unsure about whether `PreRender` is the correct place for this, perhaps overriding the setter functions to also update the shader constants would be better? And I'm not sure whether other settings like AmbientTemperature should also be covered by this or if it is intended to always be fixed.

But I do hope that I'm not wrong in expecting that at least `resolution`, `maxTemp` and `minTemp` are supposed to cause direct change in the output, otherwise I apologize. In that case I suppose I should file a bug with `gz-sim-thermal-sensor-system` instead to add some kind of call that resets the thermal camera whenever it updates its settings?

iche033, 2025-05-09:
yes your understanding of the code is correct and that the `PreRender` function is the right place to put it because `PreRender` is guaranteed to be called in the rendering thread by gazebo. We also do this in other places, e.g. [Ogre2DepthCamera](https://github.com/gazebosim/gz-rendering/blob/25528d26e8bcc22baba6476d432028dcf8237bff/ogre2/src/Ogre2DepthCamera.cc#L198-L204). Yes we should at least update `resolution`, `maxTemp` and `minTemp`. 

As for `ambient`, we can also  set it too in `PreRender` but I noticed that is also used by the [quad pass](https://github.com/gazebosim/gz-rendering/blob/25528d26e8bcc22baba6476d432028dcf8237bff/ogre2/src/Ogre2ThermalCamera.cc#L1019) (to initialize the texture values before each operation) so we may also need to update the quad pass definition programmatically but that can be a todo item.

would you like to create a PR with your proposed fix?

mjacobse, 2025-05-11:
Great, thanks for confirming. Sure, if you think it's the right way to fix this issue I can do a PR: #1138

Unfortunately I seem to be unable to get the fix to work for Ogre1, I might need some help with that.

mjacobse, 2025-05-14:
Should we close this or leave it open for Ogre1 and potentially other settings?

iche033, 2025-05-14:
I'll track ogre1.x specific thermal camera tasks in https://github.com/gazebosim/gz-rendering/issues/1140. Updated the description there to include adding supporting for changing thermal camera settings at runtime.

Closing this issue.

commit 55fb3c55 (2025-05-13): Fix user settings of thermal camera being ignored (#1138)

Fixes #1134

Changes to thermal camera settings that can be made through the plugin
gz-sim-thermal-sensor-system would not show up in the result. While the
value of the data members was updated, the named constants in the
material shader used for rendering were not. We fix that by always
setting the shader constants in PreRender to the current values.

This might be relevant for other thermal camera settings as well, but
this focuses on min and max temperature and resolution, as those can be
modified through the gz-sim-thermal-sensor-s [... cut]
commit 42bd532f (2025-07-10): Fix user settings of thermal camera being ignored (#1138)

Fixes #1134

Changes to thermal camera settings that can be made through the plugin
gz-sim-thermal-sensor-system would not show up in the result. While the
value of the data members was updated, the named constants in the
material shader used for rendering were not. We fix that by always
setting the shader constants in PreRender to the current values.

This might be relevant for other thermal camera settings as well, but
this focuses on min and max temperature and resolution, as those can be
modified through the gz-sim-thermal-sensor-s [... cut]
commit 7d311d9a (2025-07-11): Fix user settings of thermal camera being ignored (#1138)

Fixes #1134

Changes to thermal camera settings that can be made through the plugin
gz-sim-thermal-sensor-system would not show up in the result. While the
value of the data members was updated, the named constants in the
material shader used for rendering were not. We fix that by always
setting the shader constants in PreRender to the current values.

This might be relevant for other thermal camera settings as well, but
this focuses on min and max temperature and resolution, as those can be
modified through the gz-sim-thermal-sensor-s [... cut]
closed 2025-05-14 by iche033